ghdl/ghdl #380: Error compiling UVVM

qarlosalberto, 2017-07-04:
I have an error with AXI-Lite and AXI-Stream compiling UVVM libraries

`cruiz@cruiz-pc:~/Descargas$ /usr/local/lib/ghdl/vendors/compile-uvvm.sh --uvvm-vip --src ./UVVM_All/  --out ./pepe`

`******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-35-g92973ee) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /home/cruiz/Descargas/pepe/bitvis_vip_axistream/v08/
Command line:
/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/5.4.0/ghdl1 -fexplicit -frelaxed-rules --no-vital-checks --warn-binding --mb-comments --std=08 --work=bitvis_vip_axistream -P/home/cruiz/Descargas/./pepe/ -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -quiet -o vvc_methods_pkg.s /home/cruiz/Descargas/UVVM_All/bitvis_vip_axistream/src/vvc_methods_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Call stack traceback locations:
0x6757f4 0x68a3d6 0x68a43a 0x68c6aa 0x73c395 0x73dc55 0x73b80f 0x743312 0x7438ef 0x747c1e 0x749743 0x69d5d3 0x71e710 0x668097 0x64307e 0xab0dbf 0x63f990 0x648278 0x6680e9 0x63f185 0x7f5fd8a3382e 0x640a07 0xfffffffffffffffe
******************************************************************
`

Paebbels, 2017-07-06:
Please update your GHDL installation. There have been fixes/feature additions for UVVM since `01.03.2017+35`.

qarlosalberto, 2017-07-06:
@Paebbels I have the last GHDL:

`cruiz@cruiz-pc:~/Descargas/ghdl$ git log --pretty=format:"%h - %an, %ar : %s"
7cbb32c - Tristan Gingold, hace 4 días : Add std_logic_misc to synopsys v08.
`

Paebbels, 2017-07-06:
@qarlosalberto But your exception log shows that GHDL was build from Git commit `g92973ee`, which is 35 commits after tzhe last tagged release (01.03.2017). We should now have more then 100 commits since 01.03.2017.

Please check the file date of GHDL and confirm that your used executable is older.

qarlosalberto, 2017-07-06:
I have recompiled ghdl but the same version appears ...

```
git clone https://github.com/tgingold/ghdl.git
wget https://ftp.gnu.org/gnu/gcc/gcc-5.4.0/gcc-5.4.0.tar.gz
tar xvzf gcc-5.4.0.tar.gz
cd ghdl; ./configure --with-gcc=../gcc-5.4.0 --prefix=/usr/local; make copy-sources
cd /home/cruiz/Descargas
cd gcc-5.4.0; ./contrib/download_prerequisites; mkdir gcc-objs; cd mpc
cd /home/cruiz/Descargas
cd gcc-5.4.0/gcc-objs; ../configure --prefix=/usr/local --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath
cd /home/cruiz/Descargas
cd gcc-5.4.0/gcc-objs; make -j2; sudo make install
cd /home/cruiz/Descargas
cd ghdl; make ghdllib; sudo make install
```

Paebbels, 2017-07-06:
@qarlosalberto See line 3 in `version.ads`. It says `171` since last tag (01.03.2017). This is much newer than `35` as in your original post.

After recompilation and reinstallation, `ghdl -v` should show `171`, otherwise the build and/or installation flow is not correct.

qarlosalberto, 2017-07-06:
ahhh I saw only g5783528. Now It is:
```
cruiz@cruiz-pc:~$ ghdl -v
GHDL 0.34-dev (2017-03-01-171-g5783528) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Now I only have problems with AXI-Stream:

```
/usr/local/lib/ghdl/vendors/compile-uvvm.sh --all --src ./UVVM_All/  --out ./pepe
```

```
Compiling library 'bitvis_vip_axistream'...
Analyzing package '/home/cruiz/Descargas/UVVM_All/bitvis_vip_axistream/src/axistream_bfm_pkg.vhd'

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-171-g5783528) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /home/cruiz/Descargas/pepe/bitvis_vip_axistream/v08/
Command line:
/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/5.4.0/ghdl1 -fexplicit -frelaxed-rules --no-vital-checks --warn-binding --mb-comments --std=08 --work=bitvis_vip_axistream -P/home/cruiz/Descargas/./pepe/ -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -quiet -o axistream_bfm_pkg.s /home/cruiz/Descargas/UVVM_All/bitvis_vip_axistream/src/axistream_bfm_pkg.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: iirs_utils.adb:1084
Call stack traceback locations:
0x7fa1cdb85a81 0x6899cf 0x689958 0x713fd4 0x71945b 0x701a51 0x704f83 0x705dbd 0x71d341 0x70e335 0x71e9ea 0x6ec04e 0x72492a 0x66b9c7 0x643e0e 0xab702f 0x640730 0x649008 0x66ba19 0x63ff25 0x7fa1cd29b82e 0x6417a7 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: iirs_utils.adb:1084
Call stack traceback locations:
0x7fa1cdb85a81 0x6899cf 0x689958 0x713fd4 0x71945b 0x701a51 0x704f83 0x705dbd 0x71d341 0x70e335 0x71e9ea 0x6ec04e 0x72492a 0x66b9c7 0x643e0e 0xab702f 0x640730 0x649008 0x66ba19 0x63ff25 0x7fa1cd29b82e 0x6417a7 0xfffffffffffffffe
/usr/local/bin/ghdl: compilation error
```

tgingold, 2017-10-29:
Should now work.

closed 2017-10-29 by tgingold