ghdl/ghdl #381: No field Enum_Pos

rongcuid, 2017-07-08:
I got this message when I am developing my project, here: https://github.com/rongcuid/lots-of-subleq-cpus/commit/3fe71c165bb4e4c68d32b6ccaafae8497b243136 I don't know what it means, honestly

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /mnt/hgfs/Projects/Subeq/Upgraded-SUBLEQ/lots-of-subleq-cpus/Subleq Pipelined/
Command line:
ghdl -m --workdir=work --ieee=synopsys test_mmu_tb
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Enum_Pos
Call stack traceback locations:
0x7fc43b64a8d1 0x56539d39ff48 0x56539d41cae4 0x56539d564393 0x56539d5648ad 0x56539d56671d 0x56539d56697e 0x56539d5675c9 0x56539d56762f 0x56539d567a8f 0x56539d567dd6 0x56539d596216 0x56539d59cab0 0x56539d43ad92 0x56539d43b52e 0x56539d480dbe 0x56539d4815f2 0x56539d4816a5 0x56539d480732 0x56539d481f24 0x56539d48c5c7 0x56539d557cb6 0x56539d553e6e 0x56539d3c4fa9 0x56539d2e5496 0x7fc43aa4f2af 0x56539d2e42d8 0xfffffffffffffffe
******************************************************************
```

Paebbels, 2017-07-08:
The used GHDL version is a bit old. This version is from before we changed the version string to include the Git hash from which GHDL was build. Any chance to update GHDL in that project or to run it locally?

GHDL received a lot of bug fixes so your discovered bug might already be fixed.

tgingold, 2017-07-08:
I cannot reproduce the crash with the development version, so I suppose the issue was fixed.

Tristan.

keesj, 2018-03-24:
I have a similar crash now:

```
keesj@700z:~$ ghdl  -v
GHDL 0.36-dev (v0.35-94-gf90933be) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

test.vhd
```
library ieee;
use ieee.std_logic_1164.all;

entity can_rx is
  port (
          demo     : in std_logic;
          status : out std_logic_vector (31 downto 0)
  );
end can_rx;

architecture rtl of can_rx is
begin
 status(0) <= (0=>'0', others => '0') when demo = '1' else (0=>'1', others => '0');
end rtl;`
```

when normally compiled I get the following message:
```
keesj@700z:~/projects/fpga/test$ ghdl -a test.vhd 
test.vhd:13:15:error: type enumeration subtype "std_logic" is not composite
test.vhd:13:60:error: type enumeration subtype "std_logic" is not composite
```

but when compiling with the following flags I get a crash
```
keesj@700z:~/projects/fpga/test$ ghdl -a --ieee=synopsys --std=08 test.vhd 
test.vhd:13:15:error: type the anonymous enumeration subtype defined at ../../src/ieee2008/std_logic_1164.vhdl:89:32 is not composite

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-94-gf90933be) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: x86_64-pc-linux-gnu
In directory: /home/keesj/projects/fpga/test/
Command line:
ghdl -a --ieee=synopsys --std=08 test.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Enum_Pos
Call stack traceback locations:
0x6ccecf 0x4bb3e1 0x58173e 0x531909 0x531e2d 0x534de9 0x5351e9 0x56451f 0x56aca3 0x5288a6 0x699cf8 0x68b278 0x6b3039 0x4062cb 0x7fb9a4d1b1bf 0x4050a8 0xfffffffffffffffe
******************************************************************
```

Paebbels, 2018-03-25:
Please create a new issue on GitHub. You can link to this (solved) issue if you see a relationship.

Paebbels, 2018-04-05:
@tgingold  
Should this issue be closed? Your commit message contains a hint to issue #381, but no interpretable command like `fixes #381` or `fix for #381` like in older commits.

closed 2018-04-06 by tgingold