ghdl/ghdl #382: no output binaries and error trying -r

zepatrik, 2017-07-09:
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-171-g5783528) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: ********
Command line:
ghdl -r tb_demo
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: grt-signals.adb:1906
Call stack traceback locations:
0x7f695bc50a81 0x47d69a 0x474a72 0x47590f 0x475b1d 0x475c95 0x4099dd 0x489db5 0x66c99a 0x672406 0x6710a8 0x4ec8b6 0x40960d 0x7f695b45982e 0x408427 0xfffffffffffffffe
```
Only a work-obj93.cf file is written, no binary file at all though I'm on Linux.

Paebbels, 2017-07-09:
What kind of backend is used?
(The mcode backend doesn't write an executable.)

zepatrik, 2017-07-09:
I actually just pulled the latest master, ran `make install` and that's it.

Paebbels, 2017-07-09:
Executing just

```Bash
make
sudo make install
```

This builds GHDL with mcode backend. For other backends, you need to run one of the `configure` commands if you want to use another backend.

However, GHDL shouldn't exit with an exception. Can you submit your VHDL testcase as a reproducer?

zepatrik, 2017-07-09:
[vhdl_demo1_9003235.zip](https://github.com/tgingold/ghdl/files/1133935/vhdl_demo1_9003235.zip)
I'm now trying to install it with gcc as a backend.

Paebbels, 2017-07-09:
It might be easier to build GHDL with LLVM backend. But such errors usually occur independent of the chosen backend.

zepatrik, 2017-07-09:
A friend of mine tried this on a Windows machine and the error was not reproducible, but I get it all the time.

tgingold, 2017-07-14:
I was able to reproduce a crash on master and I am fixing it.
As a workaround, limit the simulation time using --stop-time=300sec

commit e0ece4a6 (2017-07-14): Detect wait timeout overflow.

Fix #382
closed 2017-07-14 by tgingold