ghdl/ghdl #383: Exception Constraint error and library warnings

Tombhunter, 2017-07-12:
`..\..\v87\synopsys\std_logic_arith.vhd:391:14:warning: function "left_signed_arg" is never referenced`
`..\..\v87\synopsys\std_logic_arith.vhd:400:14:warning: function "left_unsigned_arg" is never referenced`
`..\..\v87\synopsys\std_logic_arith.vhd:409:14:warning: function "mult_signed_arg" is never referenced`
`..\..\v87\synopsys\std_logic_arith.vhd:418:14:warning: function "mult_unsigned_arg" is never referenced`
`..\..\v87\synopsys\std_logic_arith.vhd:1258:14:warning: function "unsigned_return_boolean" is never referenced`
`..\..\v87\synopsys\std_logic_arith.vhd:1266:14:warning:function "signed_return_boolean" is never referenced`
`FifoSyncCompOriginal.vhd:57:48:warning: universal integer bound must be numeric literal or attribute`
`FifoSyncCompPipelined.vhd:61:48:warning: universal integer bound must be numeric literal or attribute`

`******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2014 (20140331)
In directory: C:\VHDL\Ap20Fbd_C02Cld.head\Root\Libs\Fifo\Src\
Command line:
C:\HwDev\ghdl-0.33\bin\ghdl.exe -m -fexplicit --ieee=synopsys -PC:\VHDL\Ap20Fbd_C02Cld.head\Root\Libs\AslGen\Src -PC:\HwDev\XILINX\14.7\Vivado\unisim\v93 -PC:\VHDL\Ap20Fbd_C02Cld.head\Root\Libs\Fifo\Src fifosync_tb
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: files_map.adb:731 index check failed ******************************************************************`

I've been trying to compile and run this testbench for a long time now, with various errors. I'm very new to both GHDL and VHDL so it may be something that I've been doing wrong but I can't seem to fix this

Another error that I've been getting whilst trying to compile something similar has been this

`C:\VHDL\Ap20Fbd_C02Cld.head\Root\Libs\Fifo\Src>ghdl -r -fexplicit --ieee=synopsys -PC:\VHDL\Ap20Fbd_C02Cld.head\Root\Libs\AslGen\Src -PC:\HwDev\XILINX\14.7\Vivado\unisim\v93 -PC:\VHDL\Ap20Fbd_C02Cld.head\Root\Libs\Fifo\Src fifosync_tb
C:\HwDev\ghdl-0.33\bin\ghdl.exe:error: overflow detected
C:\HwDev\ghdl-0.33\bin\ghdl.exe:error: error during elaboration`

tgingold, 2017-07-12:
You are using a very old version of GHDL.  This issue may have been fixed since that version.

Do you have a reproducer ?

Tombhunter, 2017-07-13:
I have been trying to compile a 0.34 version of GHDL as per your suggestion, and am using the windows mcode powershell script but it says it cannot find the file "grt-backtraces-impl.ads", I am also not able to locate it in any of both the files I have downloaded and on your master repository. It says in the .gitignore file that it is "Generated files on Linux" but does this mean I cannot use it?

Tombhunter, 2017-07-14:
recompiled and run it under 0.34 and with a few tweaks it ran nicely, sorry for wasting your time :)

closed 2017-07-14 by Tombhunter

tgingold, 2017-07-15:
On 14/07/17 18:23, Tombhunter wrote:
> recompiled and run it under 0.34 and with a few tweaks it ran nicely, 
> sorry for wasting your time :)

No problems.  Great it now works nicely.