ghdl/ghdl #385: Question: Compile time code generation of packages for faster startup time?

kraigher, 2017-07-17:
I am using the latest GHDL mcode release on Windows and I notice that the startup time of a simulation becomes longer and longer for every new package that is being used even if the test bench itself is small and does not perform many operations. I suspect that GHDL is doing some code generation every time the simulation starts for each package contributing to the time increase. 

The NVC simulator has a `--codegen` flag which allows a package to be converted to native code already during compile time allowing a faster simulation startup. When running many small simulations using complex packages code generation for packages once and for all becomes a net win for the total runtime.
When using VUnit and GHDL this use case is common.

I do not have any knowledge of the software architecture of GHDL so pardon if I got something wrong.
Is it possible to improve in this area?

tgingold, 2017-07-18:
Yes, with mcode, ghdl generates code for every simulation.
But: it only generates code that is needed, and code generation is very fast (I think it is faster to generate code than to read it from disk).

Do you have the same speed issue on linux ?  Maybe the work library is getting very large and you hit Windows filesystem performance limitations.
Maybe ghdl could avoid some disk accesses too.

kraigher, 2017-07-18:
I have a test bench that does nothing but start and stop where it takes over 12 seconds for the `--elab-run` command to finish on my laptop with a Core i5 6300U with SSD-drive.

It seems to correlate with how many packages I use not how large the work library is. When I comment out use clauses without changing the work library contents I see a speedup down to under 1 second.

Interesting to note is that when instantiating a submodule in the test bench that uses a lot of packages it adds significantly to the to run time event though everything with the architecture of the submodule has been commented out except for the port list and use-clauses.

tgingold, 2017-07-18:
On 18/07/17 06:40, kraigher wrote:
> I have a test bench that does nothing but start and stop where it takes
> over 12 seconds for the |--elab-run| command to finish on my laptop with
> a Core i5 6300U with SSD-drive.
>
> It seems to [...]

Yes, but it may not be only code generation time, but parse time too.

> Interesting to note is that when instantiating a submodule in the test 
> bench that uses a lot of packages it adds significantly to the to run 
> time event though everything with the architecture of the submodule has 
> been commented out except for the port list and use-clauses.

Can you share this test bench so that I can investigate ?

(Usually, ghdl is compiled with checks/assertion enabled, so there is
a large speed-up margin).

Tristan.

kraigher, 2017-07-18:
Yes I have the test bench on GitHub. Should I make a shell script for the test bench or can you use it through VUnit?

kraigher, 2017-07-18:
I have pushed the `ghdl_performance` branch to the VUnit repo:
https://github.com/vunit/vunit/tree/ghdl_performance/

It contains `ghdl_performance.sh` which will compile and run the test bench. Actually you notice even compile times becomes sluggish when there are many packages.

If you want to run the same thing as the shell script in VUnit use:
```
VUNIT_GHDL_PATH=path_to_ghdl/bin/ VUNIT_SIMULATOR=ghdl PYTHONPATH=. python vunit/vhdl/bfm/run.py "vunit_lib.tb_axi_stream.test read before write" --log-level=debug
```

tgingold, 2017-07-18:
A shell script would be simpler if I want to add your test bench in the 
test suite.

Envoyé avec AquaMail pour Android
http://www.aqua-mail.com

kraigher, 2017-07-18:
Yes I provided a shell script in my comment above.

kraigher, 2017-07-18:
Oh and remember to also check out submodules when checking out the VUnit repo otherwise the osvvm files will not be available.
```
git submodule update --init --recursive
```

tgingold, 2017-07-19:
I can reproduce the issue.
Most of the time is spent in the "Is_Obsolete" function: if this function is simply "return False" then execution time goes from 23sec to less than 1sec.

kraigher, 2017-07-19:
Good, always a nice situation when a single function can be optimized for a 10-20x speedup of the whole program. I rebuild GHDL-mcode and I can confirm that execution time is now back in the ~1s range. (Test on Ubuntu 16.04 LTS now)
Found any time thieves in there that can be removed?

tgingold, 2017-07-19:
Yes, I am testing a patch.  Stay tuned!

kraigher, 2017-07-19:
Looks like simple memoization of the Is_Obsolete call will reduce the time complexity significantly. When deeply nested hierarchies there is probably a lot of duplicate calls to Is_Obsolete. This is a common pattern with recursive tree operations.

closed 2017-07-20 by tgingold

kraigher, 2017-07-20:
Thanks now I am down to about 0.8 seconds per test case. Any chance it can go even lower? ;-)

tgingold, 2017-07-21:
> Thanks now I am down to about 0.8 seconds per test case. Any chance
> it can go even lower? ;-)

Sure, but note that you are mainly analyze time bound and not simulation time bound.

If you compile ghdl without checks, it is about 2x faster (in your case).
Then you can try to do parallel simulation:
* import (using ghdl -i) all the source files at once (if possible)
* run parallel simulation (using ghdl -c -r)

Tristan.

kraigher, 2017-07-21:
Actually I am not analyze bound since in the real case I have 80-100 such 0.8 ms tests and incremental compile which means that for each iteration only a few files are recompiled. I also typically run 4-8 tests in parallel as well.

kraigher, 2017-07-21:
A common use case is to edit DUT and its testbench resulting in 2 recompiles and then re-run 10-20 tests.

kraigher, 2017-07-21:
Anyway it is a big improvement the way it is now so I am happy.

commit feb9de82 (2017-07-20): libraries: mark unit obsolete to reduce the number of obsolete checks.

Fix #385