ghdl/ghdl #386: Performance speed up

BJacksonOST, 2017-07-19:
Does anyone know of any GCC options I could use to speed up simulation runtime? Currently, I'm using the latest GHDL release (0.34-dev) and the GCC version 4.9.4. I installed it using the "GHDL with mcode backend build on GNU/Linux with GCC/GNAT" instructions from the user manual. My top-level design contains many VHDL component and is very hierarchical. I presume this make runtime is slow. Currently, I'm using the GCC option -Ofast, but that barely speeds up the runtime. Suggestions.

Paebbels, 2017-07-19:
You could use another backend. e.g. the LLVM backend.

tgingold, 2017-07-20:
It is possible that you are using a particular construct that is not well handled/optimized by ghdl.
Can you post a link to your design ?  Or can you send it to me privately ?

BJacksonOST, 2017-07-24:
Unfortunately, I'm unable to do so because it's a proprietary design. In 
addition, this particular design contains just over 200 VHDL files in a 
hierarchical structure! For what it's worth, it runs well with Cadence 
NCSIM.

BJacksonOST, 2017-07-24:
I tried the LLVM backend and I received the below error. Thoughts?

Paebbels, 2017-07-24:
@BJacksonOST Sorry, but there is no error message in your post.

BJacksonOST, 2017-07-24:
Here's the error...

gcc: error: /usr/lib64/ghdl/llvm/v93/std/std_standard.o: No such file or 
directory
ghdl-llvm:error: compilation error

tgingold, 2017-07-26:
This is typically an incorrect installation. How did you build ghdl with llvm ?

tgingold, 2017-08-16:
I close this ticket, but feel free to answer and reopen.

closed 2017-08-16 by tgingold

BJacksonOST, 2017-08-16:
I'm still having performance speed issues, but you could close this 
issue. Thanks for your help.