ghdl/ghdl #388: Homographs in same declarative region

vicencb, 2017-07-22:
Hello,
the following code:
```vhdl
entity e is end entity;
architecture a of e is
  type enu is (one, two);
  function one return enu is begin
    return two;
  end function;
begin
end architecture;
```
is invalid, but `ghdl` does not report any error or warning.
The two `one` declarations are homographs and both occur in the same declarative region, which shall not be allowed.

Tested with 866e24bc5b163da2aad0f6bb5f63b7022cc1fe5c.

Regards,
&nbsp;&nbsp;Vicenç.

tgingold, 2017-07-22:
Good catch!

closed 2017-07-23 by tgingold

tgingold, 2017-07-24:
This is now fixed.
But I'd like to thank you for this issue, as I was surprized.  I think this is the largest hole that has been found for a while!

vicencb, 2017-07-24:
You are welcome!
I will continue to pursue improvements in open-source.

commit f670551a (2017-07-23): is_same_profile: handle enumeration_literal and function.

Fix #388