ghdl/ghdl #391: Exception CONSTRAINT_ERROR raised

BJacksonOST, 2017-07-25:
I'm running a hierarchical design based on Mcode backend. The simulation incredibly slow. The simulation eventually does not complete and I get the below error. Does anyone have suggestions?

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: x86_64-pc-linux-gnu
In directory: /home/bjackson/gptxhw/simulation/
Command line:
ghdl -r -O2 -g --workdir=/home/bjackson/gptxhw/simulation/work_ghdl --ieee=synopsys -fexplicit -Pcw_ghdl cpu_unit_top_check --stop-time=1000ms --ieee-asserts=disable
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-processes.adb:755 overflow check failed
Call stack traceback locations:
0x45852e 0x458c26 0x458e9a 0x459088 0x4056de 0x46cdc9 0x638509 0x63db58 0x63c810 0x4cd73e 0x40537f 0x3c6081ed1b 0x4041a7

Paebbels, 2017-07-25:
Do you have the possibility to use a newer version of GHDL or to compile the latest version from sources (e.g. with LLVM backend)?
Your version shows `0.34-dev` without a Git hash (it's like a time stamp). We changed the version string months ago, so I can see your version is a bit older.

I assume the error might already be fixed by @tgingold . Otherwise we would need a VHDL reproducer file to investigate your design that caused the error.

tgingold, 2017-08-16:
I close this ticket because of lack of progress.  Feel free to reopen.

closed 2017-08-16 by tgingold

BJacksonOST, 2017-08-16:
Sorry for the delay. Yes, your suggestion did resolve my issue.