ghdl/ghdl #392: Big arrays

vicencb, 2017-07-26:
Hello,
the following code:
```vhdl
entity e is end entity;
architecture a of e is
  type ar1 is array(integer, integer) of integer;
  type ar2 is array(integer range -2e9 to 2e9, integer range -2e9 to 2e9) of integer;
  type ar3 is array(integer range -1e9 to 2e9, integer range -1e9 to 2e9) of integer;
begin
end architecture;
```
makes `ghdl` dump the following report:
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 (v0.34) [Dunoon edition]
Compiled with GNAT Version: 7.1.1 20170630
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.1.0/ghdl1 -P/usr/lib/ghdl/v93/std/ -P/usr/lib/ghdl/v93/ieee/ -quiet -o bug.s bug.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:850 overflow check failed
Call stack traceback locations:
0x6f622c 0x6f627e 0x6f94d1 0x6fa92e 0x727ce5 0x71682d 0x705a43 0x6e2b6d 0x6d789a 0x6d6577 0x6185d8 0xae760c 0x612abd 0x61e0b9 0x740769 0x6111da 0x7f2d6a5d74c8 0x614f68 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised CONSTRAINT_ERROR : trans-chap3.adb:850 overflow check failed
Call stack traceback locations:
0x6f622c 0x6f627e 0x6f94d1 0x6fa92e 0x727ce5 0x71682d 0x705a43 0x6e2b6d 0x6d789a 0x6d6577 0x6185d8 0xae760c 0x612abd 0x61e0b9 0x740769 0x6111da 0x7f2d6a5d74c8 0x614f68 0xfffffffffffffffe
ghdl:error: compilation error
```
This is triggered by both type declarations `ar1` and `ar2`. `ar3` works fine, at least on my setup.

Regards,
&nbsp;&nbsp;Vicenç.

tmeissner, 2017-07-27:
In Modelsim DE, only the first version works. The last two generate a error message when starting the simulation with the vsim command:

`** Fatal:  (vsim-3419) Array with element size 4 and index range -1000000000 to 2000000000 is too large`

Paebbels, 2017-07-27:
@tmeissner This is a 32-bit ModelSim, right?

vicencb, 2017-07-27:
Currently this is failing at analysis time.
My expectation was that it worked fine all the way from analysis to execution, because this is only a type declaration.
If there was a declared object of such a type, then I would expect it to always work fine at analysis and elaboration and also expected it to work fine during execution only if enough memory was available.
Maybe I am expecting too much...

The behaviour of modelsim seems a bit weird, as the first one is the biggest array of them all.
In any case, it fails at execution time, almost meeting my expectations, only if there was an object of such a type.

tmeissner, 2017-07-27:
@Paebbels Yes. I will try it with Questa (64 bit Kernel) tomorrow, if interested.

@vicencb I only defined the bare types, without declaring only object using it.

Paebbels, 2017-07-27:
@vicencb 
You are right, the pure type declaration should be of no harm to any system. A 32-bit system might point out (with a warning) that an object of that type might not be created due to 32-bit memory limitations. Is can be reported because you used `integer` instead of `integer range <>`.

@tmeissner 
Yes I would like to see:
1. if a 64-bit system behaves different
2. if it allocates the object (if you add the appropriate lines).
   * as a variable, e.g. in a process
   * as a signal

tmeissner, 2017-07-28:
With Questa:

Only Type definitions:

* ar1 works
* ar2 gives following error when calling the vsim command (curiosly the error is reported for a completely other source code line):
`** Fatal: (vsim-3419) Array with element size 17179869184 and index range -2147483648 to 2147483647 (null array) is too large`
* ar3 shows the same error with the same index range, which is strange.

If declaring signals using the tyes, you get a compile error for each declaration line:

** (vcom-1311) Total array memory limit exceeded

So, when declaring signals using the types, you get an compile error early, but if you only define the types, you get the range error late during start of the simulation. Strange.

Paebbels, 2017-07-28:
@tmeissner Do you have access to Mentor to report these issues?

commit 4ebaa72d (2017-07-29): Translation: consider big arrays as non-static.

Fix #392
closed 2017-07-29 by tgingold