ghdl/ghdl #394: GHDL output unsynchronized between report and write(OUTPUT, str)

kraigher, 2017-07-28:
```vhdl
use std.textio.all;

entity bug is
end entity;

architecture a of bug is
begin
  main : process
    procedure echo(msg : string) is
      variable l : line;
    begin
      write(l, msg);
      writeline(OUTPUT, l);
    end;

  begin
    echo("1");
    report "2";
    echo("3");
    report "4";
    wait;
  end process;
end;
```
Looks good with no output pipe
```
PATH=../install-ghdl-llvm/bin/:$PATH sh test.sh
+ ghdl -a bug.vhd
+ ghdl --elab-run bug
1
bug.vhd:18:5:@0ms:(report note): 2
3
bug.vhd:20:5:@0ms:(report note): 4
```

Looks bad:
```
PATH=../install-ghdl-llvm/bin/:$PATH sh test.sh | cat
+ ghdl -a bug.vhd
+ ghdl --elab-run bug
bug.vhd:18:5:@0ms:(report note): 2
bug.vhd:20:5:@0ms:(report note): 4
1
3
```

Looks worse with `--unbuffered``
```
PATH=../install-ghdl-llvm/bin/:$PATH sh test.sh |  cat
+ ghdl -a bug.vhd
+ ghdl --elab-run bug --unbuffered
bug.vhd:18:5:@01
ms:(report note): 2
bug.vhd3
:20:5:@0ms:(report note): 4
```

It seems GHDL writes `report` statements to stderr which cause output from report and write to be unsynchronized. When using `--unbuffered` flag and `cat` the output is randomly mixed.

This is causing me problems with VUnit since we have a logging framework which uses write(OUTPUT, "") and those prints are not correctly ordered with the report prints of GHDL. We do not have the same problem with other commercial simulators. Is there a reason why report statements go to stderr? They are not really simulator error messages but part of the defined behavior of the program that simulates the VHDL code.

When VUnit runs the simulation it pipes the output from the ghdl subprocess stdout and stderr to the same file descriptor

tgingold, 2017-07-29:
To be pedantic, this is certainly not a bug as the LRM is silent about this behaviour!

Note that:
```
ghdl -r bug --unbuffered 2>&1 | cat
1
bug.vhdl:18:5:@0ms:(report note): 2
3
bug.vhdl:20:5:@0ms:(report note): 4
```

But your main point is on the difference with other simulators.  Are they also writing assert messages on stdout ?

kraigher, 2017-07-29:
I can confirm that both Modelsim and RivieraPro writes report and OUTPUT to stdout.

kraigher, 2017-07-29:
Using GHDL `--unbuffered` in VUnit does not exhibit that the unsynchronized behavior, so it solved the problem. But it is always nice if the default behavior is good without having custom flags. I could add the `--unbuffered` flag to the ghdl simulation command in VUnit but that would force all users to have the flag, maybe that has no drawbacks in practice.

kraigher, 2017-07-29:
Seems the NVC open source VHDL simulator also writes repors to stderr but write(OUTPUT) to stdout.

tgingold, 2017-07-29:
But what about assert ?

Envoyé avec AquaMail pour Android
http://www.aqua-mail.com

kraigher, 2017-07-29:
Yes asserts are also written to stdout in Modelsim and Rivierapro, even those which cause simulation break (failure). To me as a user it seems like the most logical behavior, I do not categorize the output of my VHDL simulation.

kraigher, 2017-07-29:
Why does this work:
```
> ghdl --elab-run bug --unbuffered 2>&1 | cat
1
bug.vhd:18:5:@0ms:(report note): 2
3
bug.vhd:20:5:@0ms:(report note): 4
```

But this does not
```
bug.vhd:18:5:@0ms:(report note): 2
bug.vhd:20:5:@0ms:(report note): 4
1
3
```

The second situation is the one that is closest to VUnit, we open the simulator process with stderr piped to stdout.

kraigher, 2017-07-29:
Does unbuffered mean completely unbuffered or linebuffered? Or is line-buffering the default?

diogratia, 2017-07-30:
Macbook: ghdl --version
GHDL 0.34-dev (2017-03-02-27-g1814966) [Dunoon edition]
 Compiled with GNAT Version: GPL 2015 (20150428-49)
 llvm code generator

With bug.vhdl unmodified:

Macbook: ghdl -a bug.vhdl
Macbook: ghdl -e bug
Macbook: ghdl -r bug
1
bug.vhdl:18:5:@0ms:(report note): 2
3
bug.vhdl:20:5:@0ms:(report note): 4
Macbook:

The behavior of which you'd either have to commend to OS X or llvm.

You could also note the the FLUSH file operation provided by -2008 could be expected to be effective here it it were honored. 

The underlying issue here that assertions and byte I/O (TEXTIO) don't share the same queue (stream) and most byte I/0 functions are not flushed by default.

Buffered file I/O is a host operating system optimization while disabling involves setting the buffer size to zero (setvbuf() in POSIX) causing it to flush immediately (with the write of a line here).

I've always thought VHDL simulator could benefit for initialization files or environmental variables in which you could define such behaviors alleviating the need for use run time flags (--unbuffered). 

It would appear  the correct thing would be to standardize the simulator kernel used in a multi-HDL simulation environments, which could introduce the concept of how many streams are implied here as well as requirements for flushing and ordering. Failing that removing standard ambiguity.

Notice Tristan's  `ghdl -r bug --unbuffered 2>&1 | cat` redirects stderr to stdout introducing buffering to those assertion report statements and defining ordering. That makes a useful argument for making assertion report statement go to STD_OUTPUT. It also implies the use of "error messages" in 10.3 Assertion statement para 4 ("The report clause specifies a message string to be included in error messages generated by the assertion. ...") could be changed to specify an "assertion message" no longer implying stderr. (On the theory that standards revisions should reduce rather than introduce or maintain ambiguities).

kraigher, 2017-07-30:
The most convenient behavior when you want to capture the output of the program is to have them both to stdout. This will be true of test automation systems and GUIs built on top of GHDL as they will want synchronized output.

tgingold, 2017-08-30:
The 'like other simulators' argument is convincing. Now implemented.

commit f8d8c61c (2017-08-30): Write assert and report messages to stdout

Fix #394
closed 2017-08-30 by tgingold