ghdl/ghdl #395: create_list_of_types: cannot handle IIR_KIND_SLICE_NAME

vicencb, 2017-07-31:
Hello,
the following ambiguous (and so erroneous) code:
```vhdl
entity e is end entity;
architecture a of e is
  function f(b:integer:=0) return string is begin return "abc"; end function;
  function f               return string is begin return "def"; end function;
  subtype r is integer range 1 to 2;
begin
  assert false report "x: " & f(r) severity note;
end architecture;
```
makes `ghdl` to report:
```
create_list_of_types: cannot handle IIR_KIND_SLICE_NAME (e.vhd:7:32)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 (v0.34rc11) [Dunoon edition]
Compiled with GNAT Version: 7.1.1 20170630
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.1.0/ghdl1 -P/usr/lib/ghdl/v93/std/ -P/usr/lib/ghdl/v93/ieee/ -quiet -o e.s e.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x6998a0 0x6ae52c 0x6b1946 0x681985 0x682232 0x681bac 0x67e1dd 0x66086b 0x66194f 0x661604 0x6875b8 0x6592f3 0x6d746c 0x6d6577 0x6185d8 0xae760c 0x612abd 0x61e0b9 0x740769 0x6111da 0x7f1a743eb4c8 0x614f68 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x6998a0 0x6ae52c 0x6b1946 0x681985 0x682232 0x681bac 0x67e1dd 0x66086b 0x66194f 0x661604 0x6875b8 0x6592f3 0x6d746c 0x6d6577 0x6185d8 0xae760c 0x612abd 0x61e0b9 0x740769 0x6111da 0x7f1a743eb4c8 0x614f68 0xfffffffffffffffe
ghdl:error: compilation error
```

Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2017-07-31:
@vicencb 
Is subtype `r` required for that reproducer, because lines 3 and 4 are already ambiguous to each other. Or it crashing because of that line?

vicencb, 2017-07-31:
I have just checked, and, no, subtype r is not required for the bug reproducer:
```vhdl
entity e is end entity;
architecture a of e is
  function f(b:integer:=0) return string is begin return "abc"; end function;
  function f               return string is begin return "def"; end function;
begin
  assert false report "x: " & f(1 to 2) severity note;
end architecture;
```
Lines 3 and 4 are not ambiguous to each other. In other words, they are not homographs.
Is the function call in the assertion statement that is ambiguous.

Regards,
&nbsp;&nbsp;Vicenç.

P.S.: Note to github: vhdl syntax highlighting is broken!

Paebbels, 2017-07-31:
I remember a situation where GHDL reported a homegraph for a similar problem, because line 3 shadows line 4. It's not possible to call the function from line 4.

See here: https://github.com/OSVVM/OSVVM/issues/22

If I'm correct, the problem was cause by:

```VHDL
function F(id : integer := 1) return string;
function F(id : integer) return string;

variable V : string := "foo " & F(3);
```
so & can either be:
* string & string, or
* string & char

---------
Addendum:
Why is that not an error if line 4 can never ne called?

diogratia, 2017-08-01:
> P.S.: Note to github: vhdl syntax highlighting is broken!

The dangers of monoculture. Github uses TextMate bundles for highlighting along with others.
 
[image]

Here it's caused by the end statements being on the same line as the function subprogram specifications. 

The highlighter is flagging an 'error' where a style dictate requires an end of line prior to the end statements as a delimiter. All this despite VHDL not being a line oriented language.

```vhdl
entity e is end entity;
architecture a of e is
  function f(b:integer:=0) return string is begin return "abc"; end function;
  function f               return string is begin return "def"; 
  end function;
begin
  assert false report "x: " & f(1 to 2) severity note;
end architecture;
```
The same fix applies to TextMate. The VHDL tmbundle could use work.

Alhadis, 2017-08-01:
> The same fix applies to TextMate. The VHDL tmbundle could use work

@diogratia Perhaps you should submit a pull request to the [appropriate repository](https://github.com/textmate/vhdl.tmbundle), then.

vicencb, 2017-08-01:
@Paebbels:

1. Both functions `f` in the bug report are not homographs, so, it is not an error declaring them.
1. The OSVVM ticket you are referring to was closed because the function `GetAlertLogName` and the method `GetAlertLogName` were declared in different regions, so, it is not an error even if they are homographs.
1. Finally, the code snippet you added (which is not the same case as the OSVVM ticket), both functions `F` are homographs and are declared in the same declarative region, which is not permitted.

In this bug report the only error is the call to `f` inside the assertion statement.
It is not possible to call the second `f` unambiguously and the only way to call the first `f` is to give it a parameter, which is not.
This is the reason why the call to `f` is in error and not the declarations of `f` themselves.
I hope this explanation clarifies your doubt.

tgingold, 2017-08-03:
I have a fix for the crash.  The error message is now:

e1.vhdl:6:32: can't resolve overload for function call, slice or indexed name
e1.vhdl:6:32: possible interpretations are:
e1.vhdl:6:32: slice
e1.vhdl:6:32: slice
e1.vhdl:6:29: no function declarations for operator "&"

(Could be much better).

commit 06da7bad (2017-08-03): sem_names: fix a crash with overload on a slice.

Fix #395
closed 2017-08-03 by tgingold