ghdl/ghdl #396: Secureip compiled into a seperate library

Tombhunter, 2017-08-01:
Hey Tristan,

This isn't an issue with GHDL as such but would be nice to get your feedback on.
Using your Xilinx-ISE compile Powershell results in the creation of separate library files for SecureIP and Unisim. In the Altera software that I'm used to using, they are both compiled into the same library file which means that if I reference the Unisim library it will automatically pick up all of the SecureIP library as well. This means that all my previous code only references Unisim library which seems to mean that it doesn't bind with entities in SecureIP. Do you think that this is an issue with my code or a compilation issue with the powershell?

Thank you, 
Jamie

Paebbels, 2017-08-01:
Hello @Tombhunter,

I created the PowerShell based pre-compile scripts. Can you give an example for a failing IP core?
I can not follow your analogy with Altera, because Altera has no Unisim nor can it compile SecureIP cores.

In general most SecureIP cores can not be pre-compiled because they contain encrypted VHDL code. GHDL has no decryption subsystem nor a key for the Xilinx libraries.

Another problem might be that some vendors are not so strikt about the binding rules. If a component isn't found in the specified library, they continue searching in other libraries. GHDL can mimic this behavior by enabling the "Synthesizer rules" with `--syn-binding`.

> **--syn-binding**
> Use synthesizer rules for component binding. During elaboration, if a component is not bound to an entity using VHDL LRM rules, try to find in any known library an entity whose name is the same as the component name.
> 
>There are two key points: normal VHDL LRM rules are tried first and entities are searched only in known library. A known library is a library which has been named in your design.
> 
>This option is only useful during elaboration. 

As far as I remember, I checked the created results with the pre-compiled Unisim library for QuestaSim.

Tombhunter, 2017-08-01:
I'm afraid I can't give you any of the code until tomorrow but in the meantime I have another question. When I compile the Unisim and SecureIP libraries for ISE using your powershell (sorry for assuming it was Tristan's) and the 2008 option, I get a lot of warnings saying `result subtype of a pure function cannot have access subelements`. I have come across this warning before when trying to compile my own code but as an error rather than a warning and so was wondering what ghdl option makes it a warning? 

Also when I said Altera, I meant Active-HDL, I've been working on a project in Altera at the same time and got the two muddled up.

Paebbels, 2017-08-01:
The warning might be related to `-frelaxed-rules`. The full list of flags is defined in [compile-ise.ps1 line 162](https://github.com/tgingold/ghdl/blob/master/libraries/vendors/compile-xilinx-ise.ps1?ts=2#L162)

Tombhunter, 2017-08-02:
Just been looking through the SecureIP libraries and you're absolutely right that they have been encrypted in some form. Would there be any way of decrypting these for use in GHDL or would that it be possible to precompile it with an aldec tool and then somehow link it in to run on GHDL?

Paebbels, 2017-08-02:
For both of your questions: No.

As explained before, GHDL has no decryption feature. Even if it could decrypt IP cores, we would need support from the vendors. You can generate some Secure IP cores, with wrappers by using the IP catalog / Core Generator. It will then generate a simulation model of the core based on FPGA primitives. The generated VHDL files I tried 82 years ago) was over 8 MiB in size and killed GHDL while loading...

Created files from vendors are not binary compatible to each other.

closed 2017-08-16 by tgingold