ghdl/ghdl #398: universal expressions?

vicencb, 2017-08-14:
Hello,
this is just a question, not a bug report:

in the expression `0 = 0`, which operator is selected and why?
The obvious answer is `"="[universal_integer, universal_integer return boolean]`.
But, by means of implicit conversions, `"="[integer, integer return boolean]` is also possible.

The command `ghdl --file-to-xml` tells me which operator is selected.
The LRM should tell me why, but have not found it.

In case the expression is ambiguous (and so erroneous), then this is a ~~bug~~feature in ghdl because it is accepted as valid, but please, do not fix it, current behaviour is preferred.

Thanks,
&nbsp;&nbsp;Vicenç.

Paebbels, 2017-08-14:
All numeric literals are either of type universal integer or universal real.

> **15.5.1 General**  
> There are two classes of abstract literals: real literals and integer literals. A real literal is an abstract literal that includes a point; an integer literal is an abstract literal without a point. Real literals are the literals of the type *universal_real*. Integer literals are the literals of the type *universal_integer*.

So `0 = 0` is using `"="[universal_integer, universal_integer return boolean]`.

Expressions on universal literals are universal expressions.

> **9.5 Universal expressions**  
> A *universal_expression* is either an expression that delivers a result of type *universal_integer* or one that delivers a result of type *universal_real*.

Because almost all operators result in a concrete typed integer or real, only expressions on numeric literals can again result in a *universal_****.

Attributes can also return *universal_**** values (see e.g. `INTEGER'pos`).

I don't think an implicit conversion will happen, because no type is involved. This changes immediately when one side in the comparison becomes an object (signal, variable, constant).

Operators with *universal_**** parameter take precedence over overloads that require an implizite conversion. (That how I would interpret the following overloading on p.259 ...)

    function "**" (anonymous: universal_integer; anonymous: INTEGER) return universal_integer;
    function "**" (anonymous: INTEGER; anonymous: INTEGER) return INTEGER;

tgingold, 2017-08-15:
Implicit conversions occur only when needed.
See 9.3.6 Type conversions, and the last paragraph.

(I close the issue, but feel free to continue the discussion).

closed 2017-08-15 by tgingold

vicencb, 2017-08-15:
Ok, now I see where it is stated: Last phrase of last paragraph of 9.3.6. Thanks Tristan for pointing this out!
Up until now I thought as Paebbels that this was subject to interpretation, although the interpretation was obvious.
Thank you both for your answers.