ghdl/ghdl #399: Elaboration fails with undefined reference to __gnat_last_chance_handler (and others)

phlibi, 2017-08-15:
Hi,

I compiled GHDL 0.35-dev from source with the GCC backend (tried GCC 7.2.0 and 4.9.4) according to the guide on [RTD](http://ghdl-devfork.readthedocs.io/en/alt-new/getting/Building.html). The compilation process runs through and gives me a working ghdl executable:

~~~~
$ ghdl --version
GHDL 0.35-dev (2017-03-01-190-g0879429) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
~~~~

Analysis of VHDL files works fine, tested with the hello_world example in the [Quick start guide](http://ghdl-devfork.readthedocs.io/en/alt-new/using/QuickStartGuide.html):

~~~~
$ ghdl -a hello.vhdl
$ ll
total 56
drwxrwxr-x  2 tcburch tcburch  4096 Aug 15 11:24 ./
drwx------ 43 tcburch tcburch 12288 Aug 15 11:23 ../
-rw-rw-r--  1 tcburch tcburch  5416 Aug 15 11:24 hello.o
-rw-rw-r--  1 tcburch tcburch   380 Aug 15 11:23 hello.vhdl
-rw-rw-r--  1 tcburch tcburch   193 Aug 15 11:24 work-obj93.cf
~~~~

But when trying to elaborate the design, the linker fails with undefined references to `__gnat_last_chance_handler`, `__gnat_malloc` and `__gnat_free`:

~~~~
$ ghdl -e hello_world
/usr/local/lib/ghdl/libgrt.a(grt-lib.o): In function `grt__lib__do_report':
/home/tcburch/sw/ghdl/src/grt/grt-lib.adb:55: undefined reference to `__gnat_las
t_chance_handler'
/home/tcburch/sw/ghdl/src/grt/grt-lib.adb:57: undefined reference to `__gnat_las
t_chance_handler'
# ...
~~~~

The full output can be found here:
[ghdl_undef_ref.txt](https://github.com/tgingold/ghdl/files/1224655/ghdl_undef_ref.txt)

System info:

~~~~
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.3 LTS"
$ uname -a
Linux tcburch-ThinkPad-W510 4.4.0-91-generic #114-Ubuntu SMP Tue Aug 8 11:56:56 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
~~~~

Any hints about how to fix this?

Thanks. Best regards,
Philipp

phlibi, 2017-08-15:
UPDATE: Same problem also with LLVM 3.5 code generator.

~~~~
$ ghdl --version
GHDL 0.35-dev (2017-03-01-190-g0879429) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ ghdl -e hello_world
/usr/local/lib/ghdl/libgrt.a(grt-lib.o): In function `grt__lib__do_report':
/home/tcburch/sw/ghdl/src/grt/grt-lib.adb:55: undefined reference to `__gnat_last_chance_handler'
# ...
~~~~

Is this a problem with my GNAT installation? I did not compile it myself, but used the one from the Ubuntu repository.

phlibi, 2017-08-15:
SOLVED: After installing GNAT 5.4 from the repositories and recompiling GHDL, it now works. Strange.

~~~~
$ ghdl --version
GHDL 0.35-dev (2017-03-01-190-g0879429) [Dunoon edition]
 Compiled with GNAT Version: 5.4.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ ghdl -a hello.vhdl 
$ ghdl -e hello_world
$ ./hello_world 
Hello world!
~~~~

closed 2017-08-15 by phlibi

tgingold, 2017-08-16:
Hello,

you have closed the ticket, but if you want me to investigate, please post the build log.

Tristan.

phlibi, 2017-08-16:
Hi Tristan,

in case you mean the build log emitted by ghdl which fails, this is linked in the first post (https://github.com/tgingold/ghdl/files/1224655/ghdl_undef_ref.txt). I haven't saved the output of the various runs of configure and make during the compilation of ghdl itself and haven't access to the offending machine at the moment. A try on my other computer right now did not show this issue, here everything works fine also with GNAT 4.9. Strange. So it's probably an issue with my system and nothing you need to worry about. Keep up the great work :)

Regards,
Philipp

doppioandante, 2018-11-16:
I'm getting this with current master and the LLVM backend.
I built ghdl with:
```
CXX=clang++-3.9 ./configure --with-llvm-config=$(which llvm-config-3.9) --prefix=/usr/local
```
Elaborating some testbench gives me something like this:
```
/usr/local/lib/ghdl/libgrt.a(grt-lib.o): In function `grt__lib__do_report':
/home/enrico/apps/ghdl-master/src/grt/grt-lib.adb:55: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-lib.adb:57: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-lib.adb:59: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-lib.o): In function `__ghdl_program_error':
/home/enrico/apps/ghdl-master/src/grt/grt-lib.adb:155: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-lib.o): In function `grt__lib__hi':
/home/enrico/apps/ghdl-master/src/grt/grt-lib.adb:196: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-lib.o):/home/enrico/apps/ghdl-master/src/grt/grt-lib.adb:201: more undefined references to `__gnat_last_chance_handler' follow
/usr/local/lib/ghdl/libgrt.a(grt-options.o): In function `grt__options__decode_option':
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:329: undefined reference to `__gnat_malloc'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:331: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:331: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:331: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:336: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:334: undefined reference to `__gnat_malloc'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:334: undefined reference to `__gnat_malloc'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:340: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:350: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:353: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:354: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:357: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-options.o):/home/enrico/apps/ghdl-master/src/grt/grt-options.adb:391: more undefined references to `__gnat_last_chance_handler' follow
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o): In function `grt__wave_opt__file__start':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:79: undefined reference to `__gnat_malloc'
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o): In function `grt__wave_opt__file__parse_file':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:110: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:111: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:116: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:118: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:119: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o):/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:119: more undefined references to `__gnat_last_chance_handler' follow
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o): In function `grt__wave_opt__file__initialize_tree':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:224: undefined reference to `__gnat_malloc'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:225: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:225: undefined reference to `__gnat_malloc'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:226: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o): In function `grt__wave_opt__file__tree_is_empty':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:232: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:233: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o): In function `grt__wave_opt__file__parse_path':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:247: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:248: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o):/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:254: more undefined references to `__gnat_last_chance_handler' follow
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-file.o): In function `grt__wave_opt__file__update_tree':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:349: undefined reference to `__gnat_malloc'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-file.adb:350: undefined reference to `__gnat_last_chance_handler'
[... cut ...]
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-design.adb:195: undefined reference to `__gnat_last_chance_handler'
/usr/local/lib/ghdl/libgrt.a(grt-wave_opt-design.o): In function `grt__wave_opt__design__check_sub_tree_if_all_found':
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-design.adb:211: undefined reference to `__gnat_last_chance_handler'
/home/enrico/apps/ghdl-master/src/grt/grt-wave_opt-design.adb:218: undefined reference to `__gnat_last_chance_handler'
collect2: error: ld returned 1 exit status
ghdl:error: compilation error
```

Could this be reopened?

doppioandante, 2018-11-16:
Ok, I have an idea of what the underlying problem may be:

I put CXX=clang++-3.9 before configuring because I noticed that otherwise the Makefile would not work.
This isn't actually the problem, it should work without specifying CXX, but `src/ortho/llvm4-nodebug/Makefile` contains two hardcoded variables:
```
CXX=clang++
LLVM_CONFIG=llvm-config
```
I don't have such binaries on my system, but I do have `clang++-3.9` and `llvm-config-3.9`.
Weirdly enough, changing them by hand doesn't seem to work.

doppioandante, 2018-11-17:
I'll paste you the ./configure, make and make install sequence.
Some notes: apparently gcc is used when doing make, I'm not sure whether this should be okay...
If you scroll near the bottom, you'll see that when doing `make install` it tries to use `clang++`, which is not named like so on my system.
```
enrico@ubuntu-desktop:~/apps/ghdl-master$ ./configure --with-llvm-config=$(which llvm-config-3.9) --prefix=/usr/local
Use full IEEE library
Build machine is: x86_64-linux-gnu
Debugging is not enabled with llvm 3.9.1
create pic/ subdirectory
Creating ghdl.gpr
Creating Makefile
for d in src/std src/synopsys src/openieee std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 src/ieee src/vital95 src/vital2000 src/mentor src/ieee2008 mentor/v93 ieee/v08 synopsys/v08; do \
  mkdir -p lib/ghdl/$d; \
done
Generate default_paths.ads
enrico@ubuntu-desktop:~/apps/ghdl-master$ make -j8
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.36-dev/" < src/version.in > version.tmp; \

gcc -c -g -o jumps.o /home/enrico/apps/ghdl-master/./src/grt/config/jumps.c
gcc -c -g -o times.o /home/enrico/apps/ghdl-master/./src/grt/config/times.c
gcc -c -g -o grt-cbinding.o /home/enrico/apps/ghdl-master/./src/grt/grt-cbinding.c
gcc -c -g -o grt-cvpi.o /home/enrico/apps/ghdl-master/./src/grt/grt-cvpi.c
gcc -c -g -o fstapi.o /home/enrico/apps/ghdl-master/./src/grt/fst/fstapi.c -I/home/enrico/apps/ghdl-master/./src/grt/fst
gcc -c -g -o lz4.o /home/enrico/apps/ghdl-master/./src/grt/fst/lz4.c
gcc -c -g -o fastlz.o /home/enrico/apps/ghdl-master/./src/grt/fst/fastlz.c
mkdir grt
echo "with Grt.Backtraces.Jit;" > grt-backtraces-impl.ads
echo "package Grt.Backtraces.Impl renames Grt.Backtraces.Jit;" >> grt-backtraces-impl.ads
cd grt; MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -u -c -g -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -o main.o /home/enrico/apps/ghdl-master/./src/grt/main.adb -Igrt
gcc-7 -c -I/home/enrico/apps/ghdl-master/./src/grt/ -g -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -Igrt -I- /home/enrico/apps/ghdl-master/./src/grt/main.adb
cp /home/enrico/apps/ghdl-master/./src/grt/grt.ver lib/ghdl/grt.ver
gcc -c -o vpi_thunk.o /home/enrico/apps/ghdl-master/./src/grt/vpi_thunk.c -fPIC -O -Wall
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
cd grt; MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -c -aI/home/enrico/apps/ghdl-master/./src/grt -aI.. \
  -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY ghdl_main  -cargs -g
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="/usr/bin/llvm-config-3.9" CXX="clang++" GNATMAKE="gnatmake" \
 all
gnatmake -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -aI./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe\
 ghdl_llvm -bargs -E -largs  
make[1]: Entering directory '/home/enrico/apps/ghdl-master'
clang++ -c `/usr/bin/llvm-config-3.9 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
gcc-7 -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -I./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ghdldrv/ghdl_llvm.adb
gcc-7 -c -I./ -I/home/enrico/apps/ghdl-master/./src/grt -I.. -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/apps/ghdl-master/src/grt/ghdl_main.adb
/bin/sh: 1: clang++: not found
src/ortho/llvm4-nodebug/Makefile:18: recipe for target 'llvm-cbindings.o' failed
make[1]: *** [llvm-cbindings.o] Error 127
make[1]: Leaving directory '/home/enrico/apps/ghdl-master'
Makefile:308: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
make: *** Waiting for unfinished jobs....
gcc-7 -c -I./ -I/home/enrico/apps/ghdl-master/./src/grt -I.. -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/apps/ghdl-master/src/grt/grt.ads
gcc-7 -c -I./ -I/home/enrico/apps/ghdl-master/./src/grt -I.. -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/apps/ghdl-master/src/grt/grt-errors.adb
gcc-7 -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -I./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ghdldrv/ghdldrv.adb
gcc-7 -c -I./ -I/home/enrico/apps/ghdl-master/./src/grt -I.. -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/apps/ghdl-master/src/grt/grt-main.adb
gcc-7 -c -I./ -I/home/enrico/apps/ghdl-master/./src/grt -I.. -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/apps/ghdl-master/src/grt/grt-options.adb
gcc-7 -c -I./ -I/home/enrico/apps/ghdl-master/./src/grt -I.. -gnatec/home/enrico/apps/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/apps/ghdl-master/src/grt/grt-rtis_binding.ads
[... cut ...]
 LLVM_CONFIG="/usr/bin/llvm-config-3.9" CXX="clang++" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/enrico/apps/ghdl-master'
clang++ -c `/usr/bin/llvm-config-3.9 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
/bin/sh: 1: clang++: not found
src/ortho/llvm4-nodebug/Makefile:18: recipe for target 'llvm-cbindings.o' failed
make[1]: *** [llvm-cbindings.o] Error 127
make[1]: Leaving directory '/home/enrico/apps/ghdl-master'
Makefile:308: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
```

tgingold, 2018-11-17:
When configuring for llvm, the default compiler is clang++.  What is the configure log when using CXX=clang++-3.9 ?

doppioandante, 2018-11-17:
Output of `configure` and `make` below. It fails at linking now.
```
enrico@ubuntu-desktop:~/apps/ghdl-master$ CXX=clang++-3.9 ./configure --with-llvm-config=$(which llvm-config-3.9) --prefix=/usr/local
Use full IEEE library
Build machine is: x86_64-linux-gnu
Debugging is not enabled with llvm 3.9.1
Creating ghdl.gpr
Creating Makefile
for d in src/std src/synopsys src/openieee std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 src/ieee src/vital95 src/vital2000 src/mentor src/ieee2008 mentor/v93 ieee/v08 synopsys/v08; do \
  mkdir -p lib/ghdl/$d; \
done
Generate default_paths.ads
enrico@ubuntu-desktop:~/apps/ghdl-master$ make -j8
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.36-dev/" < src/version.in > version.tmp; \

gcc -c -g -o jumps.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/config/jumps.c
gcc -c -g -o times.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/config/times.c
gcc -c -g -o grt-cbinding.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt-cbinding.c
gcc -c -g -o grt-cvpi.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt-cvpi.c
gcc -c -g -o fstapi.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/fst/fstapi.c -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/fst
gcc -c -g -o lz4.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/fst/lz4.c
gcc -c -g -o fastlz.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/fst/fastlz.c
cd grt; MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -c -aI/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt -aI.. \
  -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY ghdl_main  -cargs -g
gcc-7 -c -I./ -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt -I.. -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/grt/ghdl_main.adb
cd grt; MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -u -c -g -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -o main.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/main.adb -Igrt
cp /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.ver lib/ghdl/grt.ver
gcc -c -o vpi_thunk.o /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/vpi_thunk.c -fPIC -O -Wall
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gcc-7 -c -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/ -g -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -Igrt -I- /home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/main.adb
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="/usr/bin/llvm-config-3.9" CXX="clang++-3.9" GNATMAKE="gnatmake" \
 all
gnatmake -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -aI./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe\
 ghdl_llvm -bargs -E -largs  
make[1]: Entering directory '/home/enrico/.local/share/Trash/files/ghdl-master'
clang++-3.9 -c `/usr/bin/llvm-config-3.9 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
gcc-7 -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -I./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/ghdldrv/ghdl_llvm.adb
gcc-7 -c -I./ -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt -I.. -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/grt/grt.ads
gcc-7 -c -I./ -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt -I.. -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/grt/grt-errors.adb
gcc -o libghdlvpi.so vpi_thunk.o -shared -Wl,-soname,libghdlvpi.so
gcc-7 -c -I./ -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt -I.. -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/grt/grt-main.adb
gcc-7 -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -I./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/ghdldrv/ghdldrv.adb
gcc-7 -c -I./ -I/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt -I.. -gnatec/home/enrico/.local/share/Trash/files/ghdl-master/./src/grt/grt.adc -gnatdY -g -I- /home/enrico/.local/share/Trash/files/ghdl-master/src/grt/grt-options.adb
clang: warning: -Wl,--no-keep-files-mapped: 'linker' input unused
clang: warning: -Wl,--no-map-whole-files: 'linker' input unused
clang: warning: argument unused during compilation: '-fuse-ld=gold'
[... cut ...]
make -f ./libraries/Makefile.inc LIBSRC_DIR="./libraries" LIBDST_DIR="lib/ghdl" enable_openieee="false" LN="ln -s" CP="cp" MKDIR="mkdir" GHDL=/home/enrico/apps/ghdl-master/ghdl_llvm GHDL_FLAGS="--GHDL1=/home/enrico/apps/ghdl-master/ghdl1-llvm " vhdl.libs.all libs.vhdl.standard
make[1]: Entering directory '/home/enrico/.local/share/Trash/files/ghdl-master'
make[1]: *** No rule to make target '/home/enrico/apps/ghdl-master/ghdl_llvm', needed by 'lib/ghdl/std/v87/std-obj87.cf'.  Stop.
make[1]: Leaving directory '/home/enrico/.local/share/Trash/files/ghdl-master'
Makefile:298: recipe for target 'libs.vhdl.llvm' failed
make: *** [libs.vhdl.llvm] Error 2
```

tgingold, 2018-11-18:
make complains that ghdl_llvm cannot be found, but it has started to build it.
Can you try to -j8 ?  That would make the log easier to read.

doppioandante, 2018-11-29:
This is without -j8 and same configure as above:
```
enrico@ubuntu-desktop:~/apps/ghdl-master$ make
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.36-dev/" < src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="/usr/bin/llvm-config-3.9" CXX="clang++-3.9" GNATMAKE="gnatmake" \
 all
make[1]: Entering directory '/home/enrico/apps/ghdl-master'
clang++-3.9 -c `/usr/bin/llvm-config-3.9 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
clang: warning: -Wl,--no-keep-files-mapped: 'linker' input unused
clang: warning: -Wl,--no-map-whole-files: 'linker' input unused
clang: warning: argument unused during compilation: '-fuse-ld=gold'
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
1 warning generated.
gnatmake -o ghdl1-llvm -aI./src/ortho/llvm4-nodebug -aI./src/ortho \
-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe ortho_code_main -bargs -E \
-largs llvm-cbindings.o --LINK=clang++-3.9 \
 `/usr/bin/llvm-config-3.9 --ldflags --libs --system-libs`
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/ortho_code_main.adb
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-analysis.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-bitwriter.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-core.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-target.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-targetmachine.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-transforms.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/llvm-transforms-scalar.ads
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/vhdl/translate/ortho_front.adb
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/ortho/llvm4-nodebug/ortho_llvm.adb
gcc-7 -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /home/enrico/apps/ghdl-master/src/bug.adb
[... cut ...]
cd lib/ghdl/std/v93; /home/enrico/apps/ghdl-master/ghdl_llvm --bootstrap-standard --GHDL1=/home/enrico/apps/ghdl-master/ghdl1-llvm  --std=93
cd lib/ghdl/std/v08; /home/enrico/apps/ghdl-master/ghdl_llvm --bootstrap-standard --GHDL1=/home/enrico/apps/ghdl-master/ghdl1-llvm  --std=08
make[1]: Leaving directory '/home/enrico/apps/ghdl-master'
gcc -c -o vpi_thunk.o /home/enrico/apps/ghdl-master/./src/grt/vpi_thunk.c -fPIC -O -Wall
gcc -o libghdlvpi.so vpi_thunk.o -shared -Wl,-soname,libghdlvpi.so
enrico@ubuntu-desktop:~/apps/ghdl-master$ 
```

doppioandante, 2018-11-30:
It looks like it works now, maybe the recent fixes for llvm on windows did something? No need to reopen.