ghdl/ghdl #400: sem_name_soft: cannot handle IIR_KIND_SELECTED_NAME

vicencb, 2017-08-22:
Hello,
when compiling the following code with `ghdl -a --std=08 e.vhd` then ghdl crashes:
```vhdl
entity e is end entity;
architecture a of e is
  type p is protected
    function f(a:bit) return boolean;
  end protected;
  type p is protected body
    function f(a:bit) return boolean is begin return a='1'; end function;
  end protected body;
  shared variable v :p;
  component bb1 is port( s :out bit     ); end component;
  component bb2 is port( s :in  boolean ); end component;
  signal s1 :boolean;
  signal s2 :bit;
begin
  i1: component bb1 port map( v.f(s) => s1 );
  i2: component bb2 port map( s => v.f(s2) );
end architecture;
```
the report is:
```
sem_name_soft: cannot handle IIR_KIND_SELECTED_NAME (e.vhd:15:33)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 (2017-03-01-203-g06a78d22) [Dunoon edition]
Compiled with GNAT Version: 7.1.1 20170630
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1 --std=08 -P/usr/lib/ghdl/v08/std/ -P/usr/lib/ghdl/v08/ieee/ -quiet -o e.s e.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x699860 0x6af6db 0x6c4a64 0x683a3b 0x65f226 0x66187c 0x6615c4 0x687578 0x6592b3 0x6d742c 0x6d6537 0x618598 0xae78fc 0x612a6d 0x61e079 0x740729 0x61118a 0x7f4d45fe74c8 0x614f28 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x699860 0x6af6db 0x6c4a64 0x683a3b 0x65f226 0x66187c 0x6615c4 0x687578 0x6592b3 0x6d742c 0x6d6537 0x618598 0xae78fc 0x612a6d 0x61e079 0x740729 0x61118a 0x7f4d45fe74c8 0x614f28 0xfffffffffffffffe
ghdl:error: compilation error
```
I do not know if the given vhdl code is valid.

Regards,
&nbsp;&nbsp;Vicenç.

tgingold, 2017-08-24:
I am not sure we really want to support that, even if it appears to be valid...

Paebbels, 2017-08-24:
For i1:  
@vicencb: `v.f(...)` is a conversion function right?
Doesn't it need to be pure?

For i2:
It can fall back to an expression creating an intermediate signal.

tgingold, 2017-08-24:
> For i1:
> @vicencb : v.f(...) is a conversion function right?
> Doesn't it need to be pure?

It is pure ('impure' is not present).

> 
> For i2:
> It can fall back to an expression creating an intermediate signal.

Paebbels, 2017-08-24:
Could we argue that `v.f(...)` is a method and thus not a conversion **function** ?

vicencb, 2017-08-24:
On the one hand, the LRM clearly states which is the purpose of protected types, that is, provide atomic access to shared variables.
But, on the other hand, it does not disallow other uses.

Regardless of the LRM, ghdl is perfectly allowed not to support a ~~feature~~hole in the LRM that nobody will ever use.

So, if the solution is just to fix the crash and report an error message, that is fine, in my opinion.

commit cb1a9790 (2017-08-25): Handle protected function as association conversion function.

Fix #400
closed 2017-08-25 by tgingold