ghdl/ghdl #401: Building GHDL for Darwin on Travis-CI

Paebbels, 2017-08-23:
@diogratia
You mentioned once, that the next Darwin build will be the last you are creating...
Is this statement still true?

-------------------
So before you mothball all your development tools, please help us to setup a working Darwin build flow on Travis-CI. This platform supports native Mac OS builds too.

You'll find a lot of scripts in the repository to prepare, configure, build, test and deploy GHDL for Ubuntu, Fedora (and Windows). I don't know any other Mac OS developer with knowledge on how to build GHDL with mcode and LLVM on Mac OS. So please help us create such a flow and then you are release from the burden to create builds on your machine :).

Kind regards
    Patrick

diogratia, 2017-08-24:
This is probably not an appropriate forum to pursue this.

1. I don't need your permission to cease compiling and packaging releases. I'm not a ghdl development team member nor would I likely accede to your authority if I were.
2. Ceasing is motivated by obsolescence of my Mac. It'll boot Linux someday in the future, it's still a vary viable computer despite Apple's opinion to the contrary.
3. There are no secrets to building ghdl on OS X/MacOS. An mcode version is terribly easy, there's a dependency on llvm-3.8.0 or newer for an llvm version (Apple doesn't distribute llvm-config), a fix for Xcode 7 and later linking loaders. See the README in the current installer pkg, explains all per GPLv3. How an individual can build and install ghdl is contained in Tristan's BUILD.txt.
4. Using a Travis-CI  SDK requires the full Xcode App, running ghdl only requires Xcode Command Line Tools. All three are tied to operating system releases. This would all need to be more tightly bound to the ghdl source tree by a ghdl development team member.
5. I have no knowledge of Travis-CI and I don't have the resources to deal with it effectively on OS X/MacOS.
6. Using the Xcode App and it's package maker requires Xcode centric development which I don't do, it's definitely not open source friendly and requires releasing Xcode _projects_.
7. The installer packages I've produced were manual efforts using [Packages](http://s.sudre.free.fr/Software/Packages/about.html) by Stéphane Sudre. You could conceivably write your own command line tool and/or scripts, it's a matter of archives, XML files directing installation and script files for controlling target machine state.  The entire release effort is a matter of a few hours, dominated by running the testsuite and preparing a BOM  with a GUI for the package and regression testing. A testimony to how well organized ghdl builds are.
8. There's nothing sacred about the package distribution format, it could be replaced with an installation application. The idea here is to remove a dependency on the Xcode app. There's a side issue of signing software which isn't compatible in an Apple environment with open source. 
9. I've managed to have at least one spelling or grammar error in each installation package I've produced. It's personally embarrassing although the incidence rate and severity is decreasing.

tgingold, 2017-08-25:
Let's close if it is not the appropriate forum.

closed 2017-08-25 by tgingold