ghdl/ghdl #402: Defining operators for protected types

Paebbels, 2017-08-25:
According to this paragraph in the LRM, it should be allowed to define operators, wherein one operand is a shared variable.

**I see these three forms:**
* unary
  * `anyType := op ProtType`  
  For unary operators the method should look like this:
  ```VHDL
  type PT is protected
    impure function "abs" return integer;
  end protected;
  ```
* binary
  * `anyType := ProtType op anyType`
  * `anyType := anyType op ProtType`
 
  For binary operators the method should look like this:
  ```VHDL
  type PT is protected
    impure function "op"(operand : anyType) return anyType;
  end protected;
  ```

**Here is the LRM text:**

> **4.5.2 Operator overloading**
> The declaration of a function whose designator is an operator symbol is used to overload an operator. The sequence of characters of the operator symbol shall be one of the operators in the operator classes defined in 9.2.
>
> The subprogram specification of a unary operator shall have a single parameter, **unless the subprogram specification is a method (see 5.6.2) of a protected type. In this latter case, the subprogram specification shall have no parameters**. The subprogram specification of a binary operator shall have two parameters, **unless the subprogram specification is a method of a protected type, in which case, the subprogram specification shall have a single parameter**. If the subprogram specification of a binary operator has two parameters, for each use of this operator, the first parameter is associated with the left operand, and the second parameter is associated with the right operand.

**So I created a testcase for these LRM lines:**

```VHDL
entity e is
end entity;

architecture a of e is
	type PT is protected
		impure function "+"(operand : integer) return integer;
	end protected;
	
	type PT is protected body
		variable state : integer := 5;
		
		impure function "+"(operand : integer) return integer is
		begin
			state := state + operand;
			return state;
		end function;
	end protected body;
	
	shared variable SV : PT;
	
	constant C : integer := SV + 3;   -- line 20
begin
	assert FALSE report "C = " & integer'image(C) severity NOTE;
end architecture;
```

-----------------
**Test results from GHDL:**

```
C:\Temp\ghdl> ghdl.exe -a --std=08 .\ProztectedTypeOperators.vhdl
.\ProztectedTypeOperators.vhdl:20:35: no function declarations for operator "+"
```

Pros:
* GHDL doesn't crash
* GHDL accepts operators in protected types

Cons:
* The operator in a protected type doesn't seam to be visible or listed in the global list of operators.

-------------------
Side note:
I'm not sure if this syntax and behavior is intended or if it is a leftover from the VHDL-2008 development process, wherein they imagined more OOP like features.

tgingold, 2017-08-25:
I think you want to write `sv."+" (3)`.

Paebbels, 2017-08-25:
But then it's a normal function call using a special name, but no infix operator call, right?

tgingold, 2017-08-25:
Right.  I think the LRM is completely buggy on that part.

JimLewis, 2017-08-25:
I was surprised when @pabbles mentioned this to me the other day.   I remember the discussion in VHDL-2008.   Don't remember if it came in with the planned OOP or if it came from an ISAC resolution.   

Currently I am not sure what I would use it for.   I remember thinking the same thing when it was being discussed.   OTOH, I thought boolean_vector was kind of silly - until I found a use for it.

I was looking above, it would require the "" in the call too wouldn't it.

diogratia, 2017-08-27:
The bit about operators for protected types in -2008 4.5.2 is present in -2002:

[image]

and  was not present in -2000:
[image]

Also according to The 5.7 revision of the Shared Variable Language Change Specification Version 5.7 from 9/12/96 in 2.3 Declaring Protected Types:

[image]

There's no publicly available documentation (e.g. Language Change Specifications) for the -2002 revision to find who or why (Issue Request) the changes to 4.5.2 (2.3.1) were introduced. It'd be one or more participants in the working group leading the -2002 revision.

Paebbels, 2017-08-27:
I can't find any hint in IEEE 1076a-1999 (Shared Variables) for operator overloading.

Paebbels, 2017-08-27:
@Jim Do we have any documentation from 1993 or 1999 to 2002?

diogratia, 2017-08-29:
[image]

From which we can see ambiguity between 1076-2000 and 1076A-2000/D3, where the inclusion of the changes was propagated to 1076-2002.

Unfortunately the changes don't appear useful.

closed 2017-09-04 by tgingold