ghdl/ghdl #403: Time precision for VPI interface

abyszuk, 2017-08-29:
GHDL is reporting via VPI interface that simulator time precision is 1 ns. Is this really true?
I have a problem when building my testbench with cocotb, which checks for this value.
Basically it boils down to a fact that I can't declare an odd [ns] clock period, because the simulation will error out with message

> Send raised exception: Unable to accurately represent 7.5(ns) with the simulator precision of 1e-9

tgingold, 2017-08-30:
Will ps be ok ?

abyszuk, 2017-08-30:
Yes, I think it will be enough.

Paebbels, 2017-08-30:
Should this setting be a GHDL parameter like `-t 1fs` in other simulators, to set the minimum simulator time resolution?

abyszuk, 2017-08-31:
I don't have a strong preference here. Would be a nice feature (as new features are always nice ;-) ), but I don't think it's necessary.
Would this switch change anything else in how GHDL works?

Paebbels, 2017-08-31:
I don't think so, because in VHDL, GHDL always uses `fs` resolution.

VPI is a Verilog interface, so it has some limitations. Verilog has no concept of a physical time unit like VHDL. It has only a configuration to specify how to interpret a time value of `1` as ns, ps, ...

tgingold, 2017-09-04:
It looks like timeprecision was reported as 1ns, while 1ps was used...

commit 80eacd9d (2017-09-04): vpi: report timeprecision as 1ps

Fix #403
closed 2017-09-04 by tgingold