ghdl/ghdl #404: UVVM NULL access dereferenced

slaweksiluk, 2017-08-31:
I cant run the example tb for UVVM. I compiled UVVM with ghdl scripts. Below logs are for uart example but irqc_tb also results in the same error. At the beginning sources for UART and IRQC DUT are compiled as they are not included in uvvm compilation scripts. I used the same flags as in ghdl scripts.

`[lab-buddy@lab-monster-astripolska uvvm_examples]$ make uart_vvc_tb 
mkdir work/
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_uart/src/uart_pkg.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_uart/src/uart_pif_pkg.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_uart/src/uart_pif.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_uart/src/uart_core.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_uart/src/uart.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_irqc/src/irqc_pif_pkg.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_irqc/src/irqc_pif.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_irqc/src/irqc_core.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ /home/lab-buddy/git/github/UVVM_All/bitvis_irqc/src/irqc.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ uart_vvc_th.vhd	
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ uart_vvc_tb.vhd	
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ irqc_tb.vhd	
ghdl -e -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 --work=work --workdir=work/ -P../uvvm_lib/ uart_vvc_tb`

`[lab-buddy@lab-monster-astripolska uvvm_examples]$ ./uart_vvc_tb 
/home/lab-buddy/git/github/UVVM_All/uvvm_util/src/license_pkg.vhd:56:5:@0ms:(report note): 

*****************************************************************************************************
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
*****************************************************************************************************

/home/lab-buddy/git/github/UVVM_All/uvvm_util/src/license_pkg.vhd:85:7:@0ms:(report note): 

=====================================================================================================
=====================================================================================================
This info section may be turned off via C_SHOW_UVVM_UTILITY_LIBRARY_INFO in adaptations_pkg.vhd

Important Simulator setup: 
- Set simulator to break on severity 'FAILURE' 
- Set simulator transcript to a monospace font (e.g. Courier new)

UVVM Utility Library setup:
- It is recommended to go through the two powerpoint presentations provided with the download
- There is a Quick-Reference in the doc-directory
- In order to change layout or behaviour - please check the src*/adaptations_pkg.vhd
  This is intended for personal or company customization

License conditions are given in LICENSE.TXT
=====================================================================================================
=====================================================================================================

./uart_vvc_tb:error: NULL access dereferenced
./uart_vvc_tb:error: error during elaboration`

`[lab-buddy@lab-monster-astripolska uvvm_examples]$ ghdl -v
GHDL 0.35-dev (2017-03-01-218-g4936125) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
[lab-buddy@lab-monster-astripolska uvvm_examples]$ `

tgingold, 2017-09-03:
All the vhdl-2008 features used by UVVM are not yet supported.
I close this ticket as it duplicates some previous tickets.
(But thanks for trying to use ghdl!)

closed 2017-09-03 by tgingold

Paebbels, 2017-09-03:
Can you list what features are still missing?
Can we create issues this the tag `FeaReq: VHDL-2008` for each feature?

I know that @JimLewis is now working on OSVVM based verification IPs. These will use similar coding techniques like UVVM. He will further add support GHDL to his set of compile scripts for his traing classes.

It would be nice to add the missing features to support these libraries and in reverse be a VHDL-2008 simulator for these libraries.

Currently, GHDL is a great opportunity to run regression tests, because we don't require licenses for each parallel instance. Simulator vendors a working on regression test suites as well, but I think we are on a great way and we are a competitive solution not only for free and open source projects.

tgingold, 2017-09-05:
No, I don't have a list of missing features.

It looks there are several vhdl-2008 features that aren't used at all; we don't have any pressure to implement them.

tgingold, 2017-09-09:
> Can you list what features are still missing?
> Can we create issues this the tag FeaReq: VHDL-2008 for each feature?

Well, there are many missing features and some of them (if not most) are not used by anyone.
Let's create issues only when someone is asking for it.

My next objective is to fully support features used by UVVM.

> I know that @JimLewis is now working on OSVVM based verification IPs.
> These will use similar coding techniques like UVVM. He will further
> add support GHDL to his set of compile scripts for his traing
> classes.

Interesting to know.

> It would be nice to add the missing features to support these
> libraries and in reverse be a VHDL-2008 simulator for these
> libraries.
>
> Currently, GHDL is a great opportunity to run regression tests, [...]

Yes, I agree with that!

Tristan.