ghdl/ghdl #405: Report for assertions like the --psl-report option for PSL

tmeissner, 2017-09-02:
It would be nice, if you have a file which includes information about all VHDL assertions after simulation. It could be similar to the file generated for PSL with the --psl-report option.

Included items could be:
* Name of assertion
* File
* Line
* Count
* Status (passed/failed)

Maybe, the PSL report could be merged with this one. Then, the language should also be included in the report file.

An example:

```json
 { "type": "PSL",
   "directive": "assertion",
   "name": ".wishbonet(sim).i_wishbonemastere@wishbonemastere(rtl).wb_write",
   "file": "../syn/WishBoneMasterE.vhd",
   "line": 130,
   "count": 0,
   "status": "passed"},
 { "type": "VHDL",
   "directive": "assertion",
   "name": ".wishbonet(sim).i_wishbonemastere@wishbonemastere(rtl).reset_check",
   "file": "../syn/WishBoneMasterE.vhd",
   "line": 256,
   "count": 0,
   "status": "passed"}
```

Paebbels, 2017-09-02:
Collecting assertion statistics is also a requirement for VHDL-2017.

**Questions:**
* Why is the syntax used for name different from the external name syntax?
* How to count is some pass and some fail?

tmeissner, 2017-09-02:
I don't know why the name syntax is different, I took it from an example PSL coverage report. Counting passes & fails separately would be good. So you would know, if your assert statements were covered, even if they don't fail.

tgingold, 2017-09-05:
Do you need reports for both concurrent and sequential assertions or only for concurrent asserts ?
(The later would be much simpler)

tmeissner, 2017-09-05:
I'm honest, both would be good. But if concurrent ones are much simpler, you could implement the immediate ones later. Maybe we also should have a discussion if we should merge all these coverage statistics in one report file and its concrete structure. Like (but not similar) UCDB, or better: the export of all coverage information in one single (JSON, XML or whatever) file. This would've make the development of a post-simulation Analyse tool/script much more easier.