ghdl/ghdl #409: uvvm compile scripts

stfarley, 2017-09-08:
Hello I am trying to compile the uvvm libraries. When I look in the lib directory however I don't see any scripts. Do these need to be added from somewhere?

Paebbels, 2017-09-08:
Hello,

the scripts are located here: https://github.com/tgingold/ghdl/tree/master/libraries/vendors
When you install GHDL (0.34-dev and newer), these scripts will be copied into your installation directory's `lib/ghdl` directory.

What version are you using?
Did you use a pre-compiled package, archive or self-compiled GHDL?

stfarley, 2017-09-08:
Ah,
Thanks for the quick response. I am using 0.33. on Centos 7. I had to compile it.

I may upgrade.

Thanks,
Steve

stfarley, 2017-09-08:
Hello upgraded to 0.34 (Compiled from source) Ran a test sim. It seems to work OK.
Then tried to run the compile script for uvvm. I get this error.

/home/parallels/projects/uvvm/UVVM_All/uvvm_vvc_framework/src_target_dependent/td_target_support_pkg.vhd:43:28: no declaration for "t_channel"
eval_discrete_type_length: cannot handle IIR_KIND_SIMPLE_NAME (*std_standard*:1:1)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-226-g1cf3b7f) [Dunoon edition]
Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-11
Target: x86_64-redhat-linux
In directory: /home/parallels/projects/uvvm/bitvis_vip_avalon_mm/v08/
Command line:
/tools/ghdl/0.34/bin/ghdl -a -fexplicit -frelaxed-rules --no-vital-checks --warn-binding --mb-comments --std=08 -P/home/parallels/projects/uvvm/. --work=bitvis_vip_avalon_mm /home/parallels/projects/uvvm/UVVM_All/uvvm_vvc_framework/src_target_dependent/td_target_support_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Call stack traceback locations:
0x4f602e 0x52d31c 0x52c0ac 0x52c379 0x52c101 0x52c499 0x692c90 0x695429 0x696f67 0x69b6d6 0x69e576 0x6a471f 0x6a592f 0x548e6f 0x66157a 0x65edc9 0x4db0f6 0x40988a 0x7f4bd7f5eb33 0x408697 0xfffffffffffffffe
******************************************************************
Analyzing package '/home/parallels/projects/uvvm/UVVM_All/uvvm_vvc_framework/src_target_dependent/td_vvc_framework_common_methods_pkg.vhd'

Compile command was this
/tools/ghdl/0.34/lib/ghdl/vendors/compile-uvvm.sh --all --src UVVM_All

closed 2017-09-08 by stfarley

Paebbels, 2017-09-08:
UVVM uses some VHDL 2008 features that aren't fully implemented in GHDL. I suggest to open a new issue (for better bug tracking), because I don't remember an issue related to `eval_discrete_type_length`.