ghdl/ghdl #411: VHDL 2008, Library commands don't seem to work

stfarley, 2017-09-12:
I just noticed that when I compile with the --std-08 argument, the --dir and --clean commands don't seem to work. Not a big problem, just thought I would note it.

stfarley, 2017-09-12:
ghdl -v
GHDL 0.35-dev (2017-03-01-226-g1cf3b7f) [Dunoon edition]
 Compiled with GNAT Version: 4.8.5 20150623 (Red Hat 4.8.5-11
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Paebbels, 2017-09-12:
Does the `--clean` command also require `--std=08` to clean up `v08` directories instead of `v93` directories?

stfarley, 2017-09-12:
Just tried it now. dir seems to use the argument. (Perhaps a tweak to the User guide would be helpful here)

stfarley, 2017-09-12:
Strangely it is no longer generating .o files or the the simulation directory. 
ghdl -r --std=08 --work=tb tb --stop-time=1ms --wave=tb.ghw
ghdl:info: simulation stopped by --stop-time
✔ 12:26 ~/projects/smpte_259/sim [master L|…22] $ ls -al
total 3308
drwxrwxr-x. 2 stevefarley stevefarley    4096 Sep 12 12:24 .
drwxrwxr-x. 7 stevefarley stevefarley    4096 Sep 12 08:05 ..
-rw-rw-r--. 1 stevefarley stevefarley     388 Sep 12 12:24 ptp-obj08.cf
-rwxrwxr-x. 1 stevefarley stevefarley     260 Sep 12 12:23 run_sim
-rw-rw-r--. 1 stevefarley stevefarley 3364038 Sep 12 12:26 tb.ghw
-rw-rw-r--. 1 stevefarley stevefarley     172 Sep 12 12:24 tb-obj08.cf

tgingold, 2017-09-12:
Does it make sense for ghdl --dir (and ghdl --clean) to display the directory for all versions (and to remove all versions) ?

Yes, with the mcode backend, no .o file is generated.

Paebbels, 2017-09-12:
I'm not sure if e.g. `--clean` should remove all versions. Maybe it would be better to have a `--std=all` in such a case.

stfarley, 2017-09-12:
I would think that would be the default.

stlouk, 2017-10-01:
Hi stfarley, I also noticed this but I adding to the --dir command also --std=08 correctly displays the unit names etc.
Not sure about the clean command.
It seems the default is v93.
Hope this helps.