ghdl/ghdl #412: Crash on type generic

thukydides, 2017-09-12:
The following code crashes ghdl:
```
entity test is
        generic(
                type g_t_test
        );
        port(
                if_in  : in  g_t_test
        );
end entity;
```
Error output:
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 () [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /src/
Command line:
/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9.4/ghdl1 --std=08 -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -quiet -o test.s test.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: trans.adb:1426 access check failed
Call stack traceback locations:
0x62b52a 0x64aaa0 0x64abe9 0x64fc15 0x664ff0 0x62eb09 0x66735a 0x5ae387 0x58675e 0x97660f 0x978306 0x5ae3d9 0x583c55 0x7fa4fea4182e 0x584157 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
Exception name: CONSTRAINT_ERROR
Message: trans.adb:1426 access check failed
Call stack traceback locations:
0x62b52a 0x64aaa0 0x64abe9 0x64fc15 0x664ff0 0x62eb09 0x66735a 0x5ae387 0x58675e 0x97660f 0x978306 0x5ae3d9 0x583c55 0x7fa4fea4182e 0x584157 0xfffffffffffffffe
ghdl: compilation error
```

ghdl is `1cf3b7f8` (`master`)
gcc is `2c66a7aca70` (`gcc-7-branch` a week ago)

I guess type interfaces are not supported as of now. What is the status, are you working on that? Is there a way to contribute?

tgingold, 2017-09-13:
generic type for entities are not yet supported.  Is it something that is really needed ? (just to evaluate the priority).

Yes, you can contribute: create your fork and ask for a merge. Open issues for help!

Paebbels, 2017-09-13:
For me, it would have medium priority. So it's lower than UVVM support.

But at some point in time I would like to be able to provide (and simulate with GHDL) e.g. a synthesizable FIFO implementations. This would reduce a lot of typecasts for enumeration types to and from std_logic_vector as an universal format for data exchange between entities.

I could imagine that some basics for generic types in entities is already in place, because packages support this feature.

brouhaha, 2019-07-09:
I'm willing to offer a USD 250 bounty for support of generic types for entities.

tgingold, 2019-07-10:
To support generic types on entities, the elaboration mechanism has to be reworked.  I think this is the next big project in ghdl, after synthesis.

eric-appion, 2019-07-10:
I was hoping it wouldn't be a huge amount of work, but even so, I didn't expect $250 to be commensurate with the effort. Oh well, looking forward to seeing it someday.

brouhaha, 2019-07-10:
I was hoping it wouldn't be a huge amount of work, but even so, I didn't expect $250 to be commensurate with the effort. Oh well, looking forward to seeing it someday.

ktbarrett, 2021-05-25:
Consider this a +1. Type generic entities are extremely useful for RTL development with structural VHDL: type generic FIFOs, pipes, and FFs. These elements are used extensively (probably 60% of all instantiations) in our code base. Without type generic entities, these entities are usually written to take `std_ulogic_vector` and supporting any other type requires mapping/packing the value down to an slv and unpacking it on the other side, which can be a serious amount of glue code. A simple entity declaration that shows my intended use case:

```vhdl
entity fifo is
  generic (
    type data_type;
    depth     :       natural);
  port (
    clk       : in    std_ulogic;
    rst       : in    std_ulogic;
    data_in   : in    data_type;
    valid_in  : in    std_ulogic;
    ready_out :   out std_ulogic;
    data_out  :   out data_type;
    valid_out :   out std_ulogic;
    ready_in  : in    std_ulogic);
end entity fifo;
```

ktbarrett, 2021-12-13:
Finally got some examples together. https://gist.github.com/ktbarrett/6c971d08870811dddb9edb23988e2fda

tmeissner, 2021-12-13:
Yeah, we also had a similar case in a module which includes some pipelining functionality with various different (complex) types. We can't get it through GHDL synthesis during formal verification. In the end, separate components were written, with entity ports as the only difference between them.

tgingold, 2021-12-15:
This is now synthesized with ghdl.

dalex78, 2026-08-20:
# Issue #412 — Crash on fully generic `type` generic

Analysis by an AI agent.

## High-level summary

An entity with a fully generic (unconstrained) `type` generic and a port of that type crashed GHDL with an internal `CONSTRAINT_ERROR` at `trans.adb:1426` (the gcc-backend translator, via `ghdl1 -c`).

## Investigation / current status: already fixed on master

Checked `master` for a pre-existing test first: `testsuite/synth/issue412/` already exists, but is unrelated to this GitHub issue — it's a generic FIFO synthesis test (`generic_sfifo.vhdl`), evidently reusing the number 412 from a different context. `testsuite/gna/issue412/` did not exist.

Reproduced the exact repro from the report against current `master`, on both mcode and the gcc backend (prebuilt at this repo's exact `master` commit, matching the backend originally reported): both analyze the design cleanly, no crash. This must have been fixed as part of general improvements to unconstrained generic-type handling since this was reported (originally against GHDL 0.34, 2017). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue412/` (the exact repro from the report) whose `testsuite.sh` analyzes the design and expects success. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this issue be closed ? (PR has been merged)

closed 2026-08-24 by tgingold