ghdl/ghdl #413: Crash on access of element of invalid alias

thukydides, 2017-09-12:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity e is
end entity;

architecture arch of e is
	type t_record is record
		int : integer;
	end record;

	function conversion(slv : std_ulogic_vector) return t_record is
		variable ret : t_record;
	begin
		ret.int := to_integer(unsigned(slv));
		return ret;
	end;

	signal slv : std_ulogic_vector(127 downto 0);
	alias alias_record : t_record is conversion(slv);

	signal s_int : integer := alias_record.int;
begin
end architecture;
```
With output:
```
nonobject_alias.vhdl:21:14: subtype indication shall not appear in a nonobject alias
nonobject_alias.vhdl:21:41: name of nonobject alias is not a declaration
sem_selected_name(2): cannot handle IIR_KIND_ERROR (nonobject_alias.vhdl:21:41)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 () [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /src/
Command line:
/usr/local/libexec/gcc/x86_64-unknown-linux-gnu/4.9.4/ghdl1 --std=08 -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -quiet -o nonobject_alias.s nonobject_alias.vhdl
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Call stack traceback locations:
0x5bbb14 0x6777d1 0x686ce7 0x68447d 0x68bfe2 0x68c5bf 0x66f585 0x692619 0x5e2933 0x6671b0 0x5ae387 0x58675e 0x97660f 0x978306 0x5ae3d9 0x583c55 0x7fefc89e282e 0x584157 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Call stack traceback locations:
0x5bbb14 0x6777d1 0x686ce7 0x68447d 0x68bfe2 0x68c5bf 0x66f585 0x692619 0x5e2933 0x6671b0 0x5ae387 0x58675e 0x97660f 0x978306 0x5ae3d9 0x583c55 0x7fefc89e282e 0x584157 0xfffffffffffffffe
ghdl: compilation error
```
So this code should not compile, given that alias_record is not valid. Accessing an element of it crashes.

Accessing the whole record does not crash:
```vhdl
signal s : t_record := alias_record;
```

My guess would be there is a bug in the big case statement in [Sem_Selected_Name](https://github.com/tgingold/ghdl/blob/master/src/vhdl/sem_names.adb#L2040).

thukydides, 2017-09-12:
ghdl is 1cf3b7f (master)
gcc is 2c66a7aca70 (gcc-7-branch a week ago)

Paebbels, 2017-09-13:
It's a bit strange that GHDl reports version `0.34 ()`, whereas it was compiled with a commit from `0.35-dev`.

tgingold, 2017-09-13:
You're right: it was a missing case in sem_selected_name.

commit d202c30e (2017-09-13): sem_name: avoid a crash on error on the prefix of a selected name.

Fix #413
closed 2017-09-13 by tgingold