ghdl/ghdl #415: "signal_force" equivalent for GHDL

Zileus, 2017-09-18:
Hello,

I would like to known if the "signal_force" function is available for GHDL simulator?

regards

Basile

tgingold, 2017-09-19:
No it isn't.  IIRC this is something specific to Modelsim, that has now a equivalent in the vhdl 2008 standard.

go2sh, 2017-09-19:
@tgingold  If I remember correctly VHDL-2008 added the possibility:
a) to force a signal value and
b) to reference signals via external name.
With those two features, it should be possible to mimic the modelsim signal_force feature.

Paebbels, 2017-09-19:
VHDL-2008 code:

    a <= force '1';
    -- ...
    a <= force '0';
    -- ...
    a <= release;

The expression `'1'` can be any expression or waveform.

Zileus, 2017-09-20:
Well the VHDL 93 is still considered as the reference for many companies...

Let's do the "old-day-trick", reduce the depth of all counters used in the design...

Thans everybody.

Paebbels, 2017-09-20:
@Zileus But we are talking here about simulation code. Except for Xilinx simulators, all big simulators accept VHDL-2008 code: Active-HDL, Riviera-PRO, ModelSim, QuestaSim, GHDL.

Zileus, 2017-09-20:
i agree with you. But here, I have to use VHDL-93 code for my design and testbenchs... So I don't have the choice...

go2sh, 2017-09-20:
But it would be nice to see those two features in ghdl. ;-) But I can imagine that especially the external name feature would be hard to implement.

hipolitoguzman, 2018-02-08:
@tgingold Could this be done using the VHPI interface?

Shared variables can be accesed and modified using the VHPI i/f, as this example from Yann Guidon shows: http://ygdes.com/GHDL/direct_c/

Since the shared variables appear as symbols in executable file generated by GHDL, the symbol list can be obtained (for example) by using the nm tool. With the symbol name a #define can be introduced in the C source that will allow to access the shared variable.

But the signals don't appear in the symbol list, so it is not clear to me if they could be accessed or not. ¿Can they?

tgingold, 2018-02-08:
Yes, though the VPI interface you can set the value of a signal.

This is how cocotb works.

tgingold, 2020-08-08:
force and release are now supported.

closed 2020-08-08 by tgingold