ghdl/ghdl #416: Should GHDL offer a "Language-Server-Protocol" interface?

Paebbels, 2017-09-20:
Hello everyone,

I know there is enough work on the todo lists, but I was wondering if GHDL should eventually offer a [Language-Server-Protocol](https://github.com/Microsoft/language-server-protocol) (LSP) interface. Today, I got an email from [SourceGraph](https://about.sourcegraph.com/), that version 2.0 is now online.

SourceGraph is creating a gigantic source code index across *all* languages 1) and offering a quick search through complete GitHub for when and where an identifier is used. See the [Master Plan](https://about.sourcegraph.com/plan) for more details.

Maybe, GHDL could someday help to understand and index VHDL source files on GitHub.com or locally at companies intranets.

What are your thoughts about SourceGraph, LSP and indexing VHDL source files for a global search?

> 1) JavaScript, TypeScript, Java, Go, Python, Ruby, ...

tgingold, 2017-09-22:
I think adding LSP would be a very interesting project, but not a small one.
I have to investigate which editors support it.

Paebbels, 2017-09-22:
I found [Visual Studio Code](https://code.visualstudio.com/) as a supporter. Sigasi is thinking about that idea too.

SethGower, 2018-07-15:
I realize this is a little late, but I would love it for GHDL to provide an LSP. I have just started setting up language servers on neovim on my machine, and went to find one for VHDL and couldn't find one. I absolutely love ghdl and would love it to offer an LSP

diogratia, 2018-07-15:
[VHDL-Tool](https://www.vhdltool.com/) offers a VHDL Language Server. You can download the beta for Linux now. Caveat Emptor. It's usable with [VIM and NeoVIM](https://www.vhdltool.com/vim) as well as other editors.

It takes a lot of engineering effort to apply an LSP without wasting end user effort (look at some of the issues still in VSCode, Microsoft was changing their standard for LSPs monthly not too long ago).

From the second link it looks like you'd be better off with NeoVIM than VIM.

tgingold, 2018-07-16:
That would be nice indeed, but this requires many lines of code!

kraigher, 2018-12-07:
FYI I have started a language server project for VHDL: https://github.com/kraigher/rust_hdl
It is progressing well and already provides full syntax checks and many semantic checks. Navigate to symbol and find all references will come in a few months.

pidgeon777, 2019-02-20:
I would support anything involving LSP integration with VHDL/Verilog/SystemVerilog. Thus, I absolutely vote for LSP integration into GHDL.

tgingold, 2019-07-02:
There is now https://github.com/ghdl/ghdl-language-server.git

closed 2019-07-02 by tgingold