ghdl/ghdl #417: GTKWAVE problem with array's of std_logic_vector

ellorens, 2017-09-27:
I can't display vhdl arrays with GTKWAVE,  if i use in my design one of the array types like in my example, thouse signals don't appear in my gtkwave simulation

type type_a        is array (0 to 5) of std_logic_vector(7 downto 0);
signal try : type_a

bash : python run.py --gtkwave-fmt vcd --gui

When i run that simulation, i open gtkwave gui and then i only see normal std_logic_vectors, nor 

GHDL 0.34 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2

Paebbels, 2017-09-27:
Is this a VUnit `run.py` script?

VCD files don't support these kinds of user defined datatypes because `VCD` is part of the Verilog standard, which doesn't support such features. You should use the `GHW` output format, which is *GHDL Waveform*. GHW is supported by GTKwave.

ellorens, 2017-09-28:
Ok, our problem is we can't work with the GHW waveform file, we try to generate the signal waveform file with the command
`--gtkwave-fmt ghw `
And we get nothing from it, we also know that it is not your issue but maybe you have a solution.
Thank you in advance.

Paebbels, 2017-09-28:
The only known format that supports all datatypes (and especially nested datatypes) is currently GHW.

You could flatten all your data to a 1-dimensional `std_logic_vector`. Currently, you are using a 1-dim. vector inside of another 1-dim. vector. You might also test a 2-dimensional vector, which might be supported by VCD.

closed 2017-10-05 by tgingold