ghdl/ghdl #418: Problem with ghw

qarlosalberto, 2017-09-27:
Hello,

I have a bug when I use ghw output format. 

```
grt.rtis_utils.foreach_scalar.handle_any
```

I have not problem with vcd.

```
GHDL 0.34-dev (2017-03-01-171-g5783528) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Paebbels, 2017-09-27:
Is it possible to update to `v0.34` or `0.35-dev`?
You are using a 0.34 development version.
There have been a lot of fixes since March 2017.

Where is this "error" message from?
Please provide the full exception generated by GHDL.

qarlosalberto, 2017-09-28:
Hello, I have updated GHDL. I have the same problem:

```
Running test: ada_lib_tb.axi_interconnect_tb.test_alive
Running 1 tests

Starting ada_lib_tb.axi_interconnect_tb.test_alive
/home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/lib/UVVM/uvvm_util/src/license_pkg.vhd:56:5:@0ms:(report note): 

*****************************************************************************************************
 This is a *** LICENSED PRODUCT *** as given in the LICENSE.TXT in the root directory.
*****************************************************************************************************

/home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/lib/UVVM/uvvm_util/src/license_pkg.vhd:85:7:@0ms:(report note): 

=====================================================================================================
=====================================================================================================
This info section may be turned off via C_SHOW_UVVM_UTILITY_LIBRARY_INFO in adaptations_pkg.vhd

Important Simulator setup: 
- Set simulator to break on severity 'FAILURE' 
- Set simulator transcript to a monospace font (e.g. Courier new)

UVVM Utility Library setup:
- It is recommended to go through the two powerpoint presentations provided with the download
- There is a Quick-Reference in the doc-directory
- In order to change layout or behaviour - please check the src*/adaptations_pkg.vhd
  This is intended for personal or company customization

License conditions are given in LICENSE.TXT
=====================================================================================================
=====================================================================================================

/home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/tb/ada_tb/vunit_out/test_output/7aa886de25a64ff56340d491e4ecbb8e5accf897/ghdl/axi_interconnect_tb-bench:internal error: grt.rtis_utils.foreach_scalar.handle_any
gtkwave --save=./waveform.gtkw /home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/tb/ada_tb/vunit_out/test_output/7aa886de25a64ff56340d491e4ecbb8e5accf897/ghdl/wave.ghw

GTKWave Analyzer v3.3.81 (w)1999-2017 BSI

ghw_read_base: error in section HIE
Error in ghw file '/home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/tb/ada_tb/vunit_out/test_output/7aa886de25a64ff56340d491e4ecbb8e5accf897/ghdl/wave.ghw'.
fail (P=0 S=0 F=1 T=1) ada_lib_tb.axi_interconnect_tb.test_alive (17.6 seconds)

==== Summary =====================================================
fail ada_lib_tb.axi_interconnect_tb.test_alive (17.6 seconds)
==================================================================
pass 0 of 1
fail 1 of 1
==================================================================
Total time was 17.6 seconds
Elapsed time was 17.7 seconds
==================================================================
Some failed!
```

(I use VUnit and UVVM. I have not problem with vcd)

qarlosalberto, 2017-10-04:
Solved with:
https://github.com/tgingold/ghdl/issues/427

qarlosalberto, 2017-10-04:
I knowh the real problem. GTKWave crash with .ghw files when I use:

```
  subtype ST_AXILite_32 is t_axilite_if  (
    write_address_channel (
        awaddr(31 downto 0)  ),
    write_data_channel (
        wdata(31 downto 0),
        wstrb(3 downto 0)  ),
    read_address_channel (
        araddr(31 downto 0)  ),
    read_data_channel (
        rdata(31 downto 0)  )
  );
  constant C_AXILITE_BFM_CONFIG : t_axilite_bfm_config := (
    max_wait_cycles            => 30,
    max_wait_cycles_severity   => TB_FAILURE,
    clock_period               => 6.4 ns,
    expected_response          => OKAY,
    expected_response_severity => TB_FAILURE,
    protection_setting         => UNPRIVILIGED_UNSECURE_DATA,
    num_aw_pipe_stages         => 1,
    num_w_pipe_stages          => 1,
    num_ar_pipe_stages         => 1,
    num_r_pipe_stages          => 1,
    num_b_pipe_stages          => 1,
    id_for_bfm                 => ID_BFM,
    id_for_bfm_wait            => ID_BFM_WAIT,
    id_for_bfm_poll            => ID_BFM_POLL
  );
  signal axilite_if : ST_AXILite_32;
```

It is a UVVM type.

tgingold, 2017-10-05:
You have to try with the master of gtkwave.
A patch to gtkwave was recently merged, but I am not sure if it is available in a release yet.

qarlosalberto, 2017-10-05:
I have the same problem :/

```
/home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/tb/ada_tb/vunit_out/test_output/ada_lib_tb.axi_interconnect_tb.test_start_dsp_41d0464fb780b090ceed0d84e8857bad5219bd14/ghdl/axi_interconnect_tb-bench:internal error: grt.rtis_utils.foreach_scalar.handle_any
gtkwave /home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/tb/ada_tb/vunit_out/test_output/ada_lib_tb.axi_interconnect_tb.test_start_dsp_41d0464fb780b090ceed0d84e8857bad5219bd14/ghdl/wave.ghw

GTKWave Analyzer v3.3.86 (w)1999-2017 BSI

ghw_read_base: error in section HIE
Error in ghw file '/home/cruiz/Repositorio/95515000_RESM_AIR_EDLP/tb/ada_tb/vunit_out/test_output/ada_lib_tb.axi_interconnect_tb.test_start_dsp_41d0464fb780b090ceed0d84e8857bad5219bd14/ghdl/wave.ghw'.
```

tgingold, 2017-10-05:
Ok, this really looks like a bug.

It would be nice if you could extract a reproducer.

closed 2017-11-01 by tgingold

qarlosalberto, 2018-06-26:
You can use a filter. 

In run.py

```
  ui.set_sim_option("ghdl.sim_flags"        ,["--read-wave-opt=./filter.txt"])
```

Your filter.txt:

```
/maxhold_top_tb/uut/**
/maxhold_top_tb/signal0_tb
/maxhold_top_tb/signal1_tb
```

commit bf78529d (2017-11-01): rti: handle more unbounded record

Fix #418