ghdl/ghdl #420: Bug Report

FranzForstmayr, 2017-09-28:
******************** GHDL Bug occured ****************************
Please report this bug on http://gna.org/projects/ghdl
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2014 (20140331)
In directory: C:\Users\F_Forstmayr\Documents\VHDL\PRBS_n_par\
Command line:
C:\Software\ghdl-0.33\bin\ghdl.exe -c --std=08 abc.vhd def.vhd --vcd=PRBS.vcd
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: nodes.adb:410 index check failed
******************************************************************
I get following message from ghdl. Is there a workaround against this issue?

Yours 
Franz #

Paebbels, 2017-09-28:
Hello,

GHDL `0.33` is a very old version (2 years / from Sept. 2015). Can you update to [`0.34` release](https://github.com/tgingold/ghdl/releases/tag/v0.34)? It was released several weeks ago. There were a lot of bugfixes. There is also the possibility to use the latest `0.35-dev` version.

FranzForstmayr, 2017-09-28:
Hey, 
ok, i'm using windows. Are there already final builds, or do i have to build myself?

FranzForstmayr, 2017-09-28:
ok, i'm not able to reproduce anymore.
Thanks a lot!

closed 2017-09-28 by Paebbels