ghdl/ghdl #422: ghdl -e error

rsc9421, 2017-09-28:
Hi all, I try to compile an example with ghdl in ubuntu 17, but i get some error when i put the command "ghdl -e name_of_entity" , the capture below shows this
[image]

any response would very helpful, regards

Paebbels, 2017-09-28:
What GHDL version are you using?
What Ubuntu version are you using: 17.04 or 17.10?

We fixed such a problem for Debian testing some months ago. Maybe we need to adjust the makefiles for Ubuntu 17, too. The problem is that GCC change the behavior of Position Independent Code (PIC). GHDL is generating code differently on your system than the code shipped with some precompiled GHDL parts. The linker complains about different PIC settings of the input files.

----------------------
See also: [Has GCCs -fPIC behavior changed?](https://github.com/tgingold/ghdl/issues/213)

rsc9421, 2017-09-28:
ghdl version 0.33
how to see ghdl backend?
ubuntu 17.04

Paebbels, 2017-09-28:
Use `ghdl --version`.

Version 0.33 is quite old (Sept. 2015).
Can you update to the [`0.34` release](https://github.com/tgingold/ghdl/releases/tag/v0.34) which was release several weeks ago. The mentioned issue was fixed in `0.34`. You can also build GHDL locally to get the latest 0.35 pre-release called `0.35-dev`.

rsc9421, 2017-09-29:
Yes, I compile 0.35-dev and works very fine, thanks!

closed 2017-09-29 by Paebbels