ghdl/ghdl #424: ghdl crashes due to lack of type cast

howbacha, 2017-10-02:
The following code caused the crash:
===========================
    signal bit_cnt : std_logic_vector(2 downto 0);
    signal ack_cyc_bit_cnt : unsigned(3 downto 0);

    bit_cnt <= ack_cyc_bit_cnt(ack_cyc_bit_cnt'high-1 downto 0);

Adding type cast to std_logic_vector fixed the crash:
======================================
    bit_cnt <= std_logic_vector(ack_cyc_bit_cnt(ack_cyc_bit_cnt'high-1 downto 0));

Crash report:
==========
GHDL release: GHDL 0.33 (20150921) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Command line:
ghdl -m -v --workdir=work -Pwork testMRG
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Type
Load address: 0x103d66000
Call stack traceback locations:
0x103ecefc1 0x103d8a11b 0x103e77421 0x103e53450 0x103e53536 0x103e55099 0x103e5529f 0x103e55fda 0x103e5603c 0x103e56b5d 0x103e56e40 0x103e8e781 0x103e9698c 0x103e1f2f4 0x103e03e8c 0x103e1c55a 0x103e1d845 0x103e1d993 0x103e1b782 0x103e1e3ee 0x103e27ea1 0x103e43fca 0x103e35f38 0x103d6ad0e 0x103d67a86

Paebbels, 2017-10-02:
The used version `0.33` is over 2 years old (September 2015). Lately, we [released `0.34`](https://github.com/tgingold/ghdl/releases/tag/v0.34), which contains lots of bugfixes and improvements. The current development version is `0.35-dev`.

Can you update to at least `0.34` and verify again?

howbacha, 2017-10-02:
I update to version 0.34 as suggested and this crash issue went away being replaced by the appropriate error message.

I then ran across a new crash issue when an entity has a port defined as a std_logic_vector but the component has the port defined as std_logic.  Also the instance has a std_logic signal assigned to it.

******************* GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34 (tarball) [Dunoon edition]
Compiled with GNAT Version: GPL 2015 (20150428-49)
Target: x86_64-apple-darwin13.4.0
In directory: *****
Command line:
ghdl -m -v --workdir=work -Pwork testMRG
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: no field Range_Constraint
Load address: 0x104365000
Call stack traceback locations:
0x1044f1301 0x104398b19 0x104491eba 0x104492065 0x104480923 0x104480075 0x10447afbd 0x10447be91 0x10447c1a1 0x1044ae4fc 0x1044b55a0 0x104417027 0x10441781b 0x10444097f 0x10444a9a3 0x10446a2de 0x10445b01c 0x1043731fa 0x104366a4e
******************************************************************

Howard

Brian-Drummond, 2017-10-03:
Can you add a testcase that raises this new crash?

Paebbels, 2017-10-03:
I would like to see your second crash report to be reported in a new issue. This is because the original question was about `0.33` and the latter question is about `0.34`.

howbacha, 2017-10-03:
Here’s a tar.gz file that contains the test cases (file names s/b obvious):

- compilation is done in the “simulation” sub-directory via: “source ../ghdl_commands

Howard

______________________________
Howard Baumer
howbacha@me.com

tgingold, 2017-10-04:
Did you forget the files ?

howbacha, 2017-10-04:
I attached a tar.gz file.  here it is again.  Hope your email system isn’t preventing it from going through

Howard

go2sh, 2017-10-04:
@howbacha  I dont think Github allows, what you are trying todo. :( Only certain file extensions are allowed. E.G. txt

Paebbels, 2017-10-04:
Did you send your file as an email attachment? I'm not sure if email
attachments are allowed or translated into comments with file uploads.

Please use the web interface.

howbacha, 2017-10-04:
Here's the upload
[scratch.tar.gz](https://github.com/tgingold/ghdl/files/1356514/scratch.tar.gz)

tgingold, 2017-10-04:
I can reproduce the crash.  Investigating.

tgingold, 2017-10-04:
Now fixed.  The reproducer is helpful!

commit ad0dd0fa (2017-10-04): sem_specs: do not check restrictions for incompatible associations.

Fix #424
closed 2017-10-04 by tgingold