ghdl/ghdl #427: Bug: rising_edge() does not work.

qarlosalberto, 2017-10-03:
Hello,

I'm going crazy.......

I have a problem. rising_edge() does not work with GHDL.

In this code ```r0_start``` is '0'.

```
library IEEE;
use IEEE.std_logic_1164.ALL;
use ieee.numeric_std.all;

entity fft is
 port (
    clk          : in  std_logic;                                   
    start       : in  std_logic;                              
    enable   : in  std_logic;                                   
    reset      : in  std_logic                                   
   );
end fft;
architecture Behavioral of fft is
  signal r0_start;
begin

  process(clk) begin
    if rising_edge(clk) then
      r0_start <= start;
    end if;
  end process;

end Behavioral;
```

In this code ```r0_start``` is '1'.

```

library IEEE;
use IEEE.std_logic_1164.ALL;
use ieee.numeric_std.all;

entity fft is

 port (
    clk                  : in  std_logic;                                   
    start                : in  std_logic;                              
    enable               : in  std_logic;                                   
    reset                : in  std_logic                                   
   );
end fft;

architecture Behavioral of fft is
  signal r0_start;
begin

  process(clk'event and clk='1') begin
    if rising_edge(clk) then
      r0_start <= '1';
    end if;
  end process;

end Behavioral;
```

My GHDL:

```
GHDL 0.35-dev (2017-03-01-252-g90f4fd2) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Paebbels, 2017-10-03:
Both code snippets are invalid because the signal `r0_start` has no type.

qarlosalberto, 2017-10-03:
@Paebbels, changed. I had simplified the code.

go2sh, 2017-10-03:
The bottom part is still wrong. A sensitivity list is only allowed to have signals in it. This shouldn't even compile as it is against the LRM. And you have to post a testbench which drives the signals in the fft entity to help you.

qarlosalberto, 2017-10-04:
@Paebbels @go2sh thank you.

I found the bug. The signal was assigned in two places, so there was collision. But in GTKWave the signal seemed normal. Strangely it worked with ```clk'event and clk='1'``` but no with ```rising_edge(clk)```

This also solved my problem in GTKWave: 
https://github.com/tgingold/ghdl/issues/418

```
grt.rtis_utils.foreach_scalar.handle_any
```

closed 2017-10-04 by qarlosalberto