ghdl/ghdl #430: Instance Selection when in a For Generate Loop for signal selection

Andre-Powell, 2017-10-05:
Hi All,
Would it be possible to add the ability to select which instance is required for signal selection within a for generation loop ? 

Many thanks

Andre'

tgingold, 2017-10-07:
Can you give an example ? I am not sure I understand your requirement.

Andre-Powell, 2017-10-07:
I have a design which creates a three dimensional array of processors using three nested for generate statements. 
I would like to select a specific signal for a specific instantiation, an example would be

/dut/z_sweep[1]/y_sweep[2]/x_sweep[0]/core/jump

where z_sweep, y_sweep, x_sweep are the loop ids.
I generated the entire signal list and every instantiation has no specfic id just 

/dut/z_sweep/y_sweep/x_sweep/core/jump
etc.

If the 'read_wave_file' has 
/dut/z_sweep/y_sweep/x_sweep/core/jump
then I get every single instantiation's jump signal.

I looked back through the ghdl mails prior to them stopping and one has the remark that instantiations in For Generate loops can't as yet be picked out.

If I've misunderstood something then please accept my apologies and could you advise how to select a specific instantiation in a For Generate loop.

tgingold, 2017-10-09:
I think I now understand: you are speaking about --read-wave-opt= option.  It is possible that it doesn't support the iterator.
To be implemented...

Andre-Powell, 2019-03-21:
Has this been implemented ?

tgingold, 2019-03-22:
Not yet :-(

Andre-Powell, 2024-06-01:
Hi All,
I raised this issue seven years ago, has there been any further development in this ?

All the best 

André

tgingold, 2024-06-02:
No, but contributions are always welcome!