ghdl/ghdl #435: freehdl or ghdl

stanley3384, 2017-10-07:
Hi Tristan,
    I was using ghdl a few years ago and I'm looking at picking up where I stopped.  Looking for ghdl or gvhdl all I could find was freehdl in the repositories and it looks a bit different in that it appears to be written in c++ rather than ada.  Also seems to have a built in waveform viewer rather that use gtkwave.  Before I start I'd like to be sure that freehdl is the right software to use?  If not I'll pull from github and run make etc.  Regards Ian.

Paebbels, 2017-10-07:
Hello,

GHDL also offers pre-build packages.
Have a look here: https://github.com/tgingold/ghdl/releases

GHDL has evolved over the last years, so I think it can not be compared to other solutions like FreeHDL. FreeHDL is 8 years old and doesn't support any VHDL-2008 construct. The named `gvhdl` is just a front-end script for FreeHDL.

GHDL received lots of bugfixes and new VHDL-2008 features. The number of reported exceptions in going down. And finally, companies like Sigasi are investigating how to embed GHDL into Sigasi Studio.

stanley3384, 2017-10-08:
Thanks Paebbels, I'll pull over the pre-build package or or make it from source.  Many thanks for the clarification on freehdl.  Ian.

closed 2017-10-08 by stanley3384

stanley3384, 2017-10-08:
Hi Paebbels,  I installed gnat okay, ./configure okay, make okay but "make install" failed you need "sudo make install", really minor update to README.md.
ghdl --help was okay so it must be there.

Paebbels, 2017-10-08:
Hello,

people using Linux should know when they need to call a command with or without `sudo`. It depends on where you install GHDL. E.g. if you use a prefix like `/opt/ghdl`, then `sodo` might not be required.