ghdl/ghdl #439: Overloaded generics

vicencb, 2017-10-15:
Hello Tristan,
is ghdl ready to accept bug reports related to vhdl-2008? or vhdl-2008 is still a work in progress and not mature enough?
Assuming it is ready, here is a bug report:

When compiling this code with `ghdl -a --std=08 e.vhd`
```vhdl
package p is
  generic(
    function generic_f(b:bit) return boolean;
    function generic_f(b:boolean) return bit
  );
  function f(b:bit    ) return boolean;
  function f(b:boolean) return bit    ;
end package;

package body p is
  function f(b:bit    ) return boolean is begin return generic_f(b); end function;
  function f(b:boolean) return bit     is begin return generic_f(b); end function;
end package body;

entity e is end entity;
architecture a of e is
  function f(b:bit    ) return boolean is begin return false; end function;
  function f(b:boolean) return bit     is begin return  '0' ; end function;
  package q is new work.p generic map(f,f);
begin
  assert q.f('0') report "msg2" severity note;
end architecture;
```
it reports
```
sem_name_free: cannot handle IIR_KIND_INTERFACE_FUNCTION_DECLARATION (e.vhd:4:14)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-226-g1cf3b7f8) [Dunoon edition]
Compiled with GNAT Version: 7.2.0
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1 --std=08 -P/usr/lib/ghdl/v08/std/ -P/usr/lib/ghdl/v08/ieee/ -quiet -o e.s e.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x699470 0x6ad869 0x6ada0d 0x6b26c0 0x68184f 0x67ddbd 0x660a24 0x6610be 0x684c39 0x665e8f 0x6855f3 0x686f23 0x658ed3 0x6d722c 0x6d6337 0x6181b8 0xae791c 0x61267d 0x61dc99 0x740749 0x610d9a 0x7f9a450d3f68 0x614b38 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : errorout.adb:246
Call stack traceback locations:
0x699470 0x6ad869 0x6ada0d 0x6b26c0 0x68184f 0x67ddbd 0x660a24 0x6610be 0x684c39 0x665e8f 0x6855f3 0x686f23 0x658ed3 0x6d722c 0x6d6337 0x6181b8 0xae791c 0x61267d 0x61dc99 0x740749 0x610d9a 0x7f9a450d3f68 0x614b38 0xfffffffffffffffe
ghdl:error: compilation error
```
Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2017-10-15:
If you are going to fix this issue, it might be good to look into [LCS2016-I03](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_I03). In ISAC issue 03 for VHDL-2017, we added a signature to fix the overloading problem for named associations:

VHDL-2008:
```VHDL
generic map (
  f => f,
  f => f     -- not allowed, due to uniqueness of formal names
);
```

VHDL-2017 allows an optional signature:
```VHDL
generic map (
  f [bit]     => f,
  f [boolean] => f     -- now, the formal is unique
);
```

I suspect, the implementation is not so difficult, because parsing signatures is already known to GHDL.

tgingold, 2017-10-16:
Yes, you can report bugs against implemented features of vhdl 2008.
I have a fix for this one.

commit cb252f10 (2017-10-16): Handle overload call to interface subprograms.

Fix #439
closed 2017-10-16 by tgingold