ghdl/ghdl #440: Externally visible generics

vicencb, 2017-10-15:
Hello Tristan,
the following code is invalid because interface declarations are not externally visible (except in a map aspect).
```vhdl
package p is
  generic(
    function generic_f(b:bit) return boolean
  );
end package;

entity e is end entity;
architecture a of e is
  function f(b:bit) return boolean is begin return false; end function;
  package q is new work.p generic map(f);
begin
  assert q.generic_f('0') report "msg1" severity note;
end architecture;
```
When compiling it with `ghdl -a --std=08 e.vhd` it reports:
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-226-g1cf3b7f8) [Dunoon edition]
Compiled with GNAT Version: 7.2.0
Target: x86_64-pc-linux-gnu
In directory: /tmp/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.0/ghdl1 --std=08 -P/usr/lib/ghdl/v08/std/ -P/usr/lib/ghdl/v08/ieee/ -quiet -o e.s e.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:2851
Call stack traceback locations:
0x13580f1 0x63f8c1 0x6aec8c 0x6af0c9 0x6b2745 0x6ab87f 0x6abe6d 0x6ac2bd 0x6615c7 0x6611e4 0x687198 0x658ed3 0x6d722c 0x6d6337 0x6181b8 0xae791c 0x61267d 0x61dc99 0x740749 0x610d9a 0x7f6f9c2b6f68 0x614b38 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs.adb:2851
Call stack traceback locations:
0x13580f1 0x63f8c1 0x6aec8c 0x6af0c9 0x6b2745 0x6ab87f 0x6abe6d 0x6ac2bd 0x6615c7 0x6611e4 0x687198 0x658ed3 0x6d722c 0x6d6337 0x6181b8 0xae791c 0x61267d 0x61dc99 0x740749 0x610d9a 0x7f6f9c2b6f68 0x614b38 0xfffffffffffffffe
ghdl:error: compilation error
```

Regards,
&nbsp;&nbsp;Vicenç.

tgingold, 2017-10-16:
Humm, are you sure about that ? There is no such restrictions in LRM 8.3 about package generics.
(But I would say that 4.7 implies differently).

Have you found a bug in the LRM ?

Sure, ghdl should not crash on that code, but I don't know yet how to fix it.

Paebbels, 2017-10-16:
@JimLewis, @LievenLemiengre, @NJDfan

Is a function, passed as a generic into a package visible like a function declared in the package itself?

vicencb, 2017-10-16:
I was not referring to neither 4.7 or 8.3, was referring to 12.3(p) which states visibility of selected names.

tgingold, 2017-10-16:
Be careful, as 12.3(p) is not about selected names.  It just states
that you can write: PARAM => value
where PARAM is a generic declaration.

The rule in 12.3 would be the rule f).

Tristan.

vicencb, 2017-10-16:
Ok, I should have used the term "visible by selection" instead of "selected name".
So, you are correct, as per (f), the function "generic_f" should be visible and then the provided code is valid.
4.7 talks about a declaration within a package, including both the package_header and package_declarative_part. Why would you say that it implies differently?

tgingold, 2017-10-18:
It is not clear that declarations within a package includes the interfaces.  In particular, interfaces are not immediately within the package.

12.3 f) is doesn't use the word immediately. But implying nested declarations would be silly.

(There are other issues here: you can specify an attribute for an interface object, but not for an interface subprogram!)

lievenlemiengre, 2017-10-18:
It looks like it's not clearly specified, I would consider it a bug in the spec if the generic parameters have to be stored (and made visible) in the instantiated package.

JimLewis, 2017-10-18:
I would not expect the function to be directly visible just be referencing the package.   However, I would expect that an alias in the package declarative area would make it visible.   Also for generic constants, I would expect that an external name could make their value visible.

tgingold, 2018-12-06:
I think you're right: interfaces are not considered as declaration in scope and visibility.

For external names, hummm.  Apparently the description in the LRM was written without looking at visibility.

kraigher, 2018-12-06:
Isn't an external name with `@libname.pkg.const` package prefix almost the same thing as a normal selected name `libname.pkg.const`. I think it would be illogical if the external name worked but not the normal selected name.

tgingold, 2018-12-06:
But this is clearly the intent: you can refer to an internal signal using an external name, but this is not possible with a selected name.

kraigher, 2018-12-06:
The difference makes the external name with package prefix worthwhile otherwise it would just be redundant with a selected name.

closed 2018-12-13 by tgingold