ghdl/ghdl #442: Malformed GHW VCD files

slaweksiluk, 2017-10-16:
Hi, I'm not sure if it is GTKWave or ghdl bug but i will try here first. I cannot open the GHW wave file (VCD neither) with using following record subtype signal:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity a is
end entity;

architecture bench of a is

constant WB_DAT_WIDTH	: natural := 32;

	type t_rec is record
		ack   : std_logic;
	end record t_rec;

	subtype t_sub_rec is t_rec;
	
	signal s : t_sub_rec;

	
begin

stimulus : process
begin
	wait for 1 us;
	assert false
	 report " <<<SUCCESS>>> "
	 severity failure;
	wait;
end process;

end architecture;
```

```
issue1/>  ghdl -a a.vhd
issue1/>  ghdl -e a
issue1/>  ./a --wave=w.ghw
issue1/>  ./a --wave=w.vcd
issue1/>  gtkwave w.ghw 
Gtk-Message: Failed to load module "canberra-gtk-module"

GTKWave Analyzer v3.3.80 (w)1999-2017 BSI

ghw_read_type: unknown type 38
ghw_read_base: error in section TYP
Error in ghw file 'w.ghw'.
issue1/>  gtkwave w.vcd 
Gtk-Message: Failed to load module "canberra-gtk-module"

GTKWave Analyzer v3.3.80 (w)1999-2017 BSI

Near byte 95, VCD search table NULL..is this a VCD file?
Near byte 95, Unknown VCD identifier: ''
No symbols in VCD file..is it malformed?  Exiting!
```

```
issue1/>  ghdl -v
GHDL 0.35-dev (2017-03-01-218-g4936125) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2017-10-16:
For vcd: there are no signals dumped, so gtkwave don't like it.
For ghw: I think this is fixed with a recent version of *both* gtkwave and ghdl.

slaweksiluk, 2017-10-17:
Yes, its working after recompiling with recent repo heads of both ghdl and gtkwave

closed 2017-10-17 by slaweksiluk

diogratia, 2017-10-17:
The change to support  record composites appears to have been added in gtkwave-3.3.81. 

From the Changelog:
```
3.3.81	09jun17	Added max_fsdb_trees environment variable.  
		Fixed -C option so it is persistent across new tabs.		  
		Integrated updated GHW reader code.  
```

The current version is 3.3.86 04Oct17

Don't you need a transaction after time 0 to get anything to show up in  GTKwave?

slaweksiluk, 2017-10-17:
I had used 3.3.80 previously. The reproducer is not perfect but with the 3.3.86 gtkwave is launched properly without crash.