ghdl/ghdl #443: Building with custom gnatmake

sonOfRa, 2017-10-19:
My system doesn't have a gnatmake symlink, instead it only has the gnatmake-6.3.0 executable to indicate the gcc version. The configure script actually works properly when I set ``GNATMAKE=gnatmake-6.3.0 ./configure``. The top level makefile now has ``GNATMAKE=gnatmake-6.3.0``. However all the sub-makefiles (like https://github.com/tgingold/ghdl/blob/master/src/ortho/llvm/Makefile) still assume that the ``gnatmake`` is on the path, and then the build fails. I "fixed" it manually by adjusting the Makefiles, but I'm not sure if this could possibly be fixed so this becomes unnecessary in the future.

tgingold, 2017-10-19:
I have pushed a fix.  Tell me if this isn't complete.

sonOfRa, 2017-10-20:
That looks to have done the trick. Thanks for the fast resolution!

closed 2017-10-20 by sonOfRa