ghdl/ghdl #444: Consecutive assignments within a process

mariobarbareschi, 2017-10-20:
Simulating something like

```vhdl
signal test : std_logic_vector(7 downto 0) := x"00";

...

dummy: process(...)
begin
        test <= x"ff";
        test <= x"7f" after 5 ns;
end process;
```

causes the output to be always at the second assignment statement value. I checked with other simulators and even the VHDL standard: the signal should be "ff" for 5 ns before going to "7f". Am I wrong or there is a kind of optimization that keep the signal always to the value of second assignment?

tgingold, 2017-10-20:
GHDL is correct here.  This feature is called pulse rejection.

You can get "ff" before "7f" if you use the transport mechanism.

Feel free to argue, but please quote the VHDL standard precisely.

Paebbels, 2017-10-20:
Solution with `transport`:

```VHDL
test <= x"ff";
test <= transport x"7f" after 5 ns;
```

Or you can assign it in one line:

```VHDL
test <= x"ff", x"7f" after 5 ns;
```

closed 2017-10-27 by tgingold