ghdl/ghdl #445: Install both gcc and llvm versions of ghdl

vicencb, 2017-10-21:
Hi Tristan,
I was wondering if it would be possible to install the gcc version of ghdl alongside with the llvm version.
Of course there is the obvious conflict that both versions install `.../bin/ghdl`
Other than that, there are also differences to
```
.../lib/ghdl/grt.lst
.../lib/ghdl/libgrt.a
.../lib/ghdl/v*/.../*.o
.../lib/ghdl/v*/.../*.cf
.../lib/libghdlvpi.so
```
For example, the differences to *.cf are only timestamps, so, those files could be shared among both installations.
What about the others, could they also be shared?
Or otherwise the only solution is to install both in different prefixes?

Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2017-10-21:
I have a multiple backend installation in `/opt/ghdl/0.35-dev-<backend>/`.

Are you also suggesting to have a unified ghdl binary with a backend switch?

Example:. 
`.../bin/ghdl --llvm -a file.vhdl`

vicencb, 2017-10-21:
No, I was not suggesting to have one ghdl that could switch backends dinamically.
My idea was that if most files are compatible among backends, then both installations could share the same prefix.
Then, `.../bin/ghdl` could be renamed to `.../bin/ghdl-gcc` and `.../bin/ghdl-llvm`. Also a link `.../bin/ghdl` could be there to select one of the two by default.
That will require no changes in ghdl, it is only a packaging issue.

tgingold, 2017-10-21:
All the lib/ files can be shared.
You can even analyze some files with llvm and others with gcc, they are compatible (provided the ghdl version is the same).

But of course, this is asking for troubles!

tgingold, 2017-10-21:
Is it a question of curiosity or do you have a use case ?

closed 2017-10-21 by tgingold

vicencb, 2017-10-21:
Thanks for clarifying.
I have a small use case, only applicable for testing or development.
An end user will only be interested in one version, either the gcc one or the llvm one.
But for testing it is good to have both and check for differences.
The use case is just to save disc space if those files are really equivalent.

Regards,
  Vicenç.