ghdl/ghdl #446: Filename always ends with -obj93.cf, even with std != 93

TheMassController, 2017-10-25:
Take the following series of commands:
```
ghdl -i --work=work --std=02 --workdir=work/ axi4_acp_writer.vhd axi4_acp_reader.vhd tb_axi4_acp.vhd axi4_acp.vhd
ghdl -m --work=work --std=02 --workdir=work/ tb_axi4_acp
```
Now the output file is called:
`work/work-obj93.cf`
Which seems misleading, since the standard is 02 and not 93. I tried every standard and they all give me the same output file.

Paebbels, 2017-10-25:
What GHDL version are you using?

Please use `ghdl -v`.

TheMassController, 2017-10-25:
I did delete all files related to the GHDL build. 
```
~ ghdl -v
GHDL 0.34 (2017-03-01-203-g06a78d22) [Dunoon edition]
 Compiled with GNAT Version: 7.2.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2017-10-25:
Because they are very compatible, vhdl93, 00 and 02 use -obj93.cf
However, vhdl87 or vhdl08 use a different filename for the libraries.

No real issue here.

TheMassController, 2017-10-26:
Very compatible or actually compatible? Unless the are really compatible you can get all sorts of weird bugs by assuming that a obj93.cf file has the 93 standard.

tgingold, 2017-10-26:
No, because the standard is set by --std=XX.
If you mix 93 and 02, and if you use 02 features (protected types), you will have to elaborate using vhdl-02.

Of course, vhdl-02 is not fully compatible with vhdl-93 (new reserved words). But apart that, it consists of new features.

This is not the case for vhdl-08 whose ieee packages are rather different from vhdl-93.

TheMassController, 2017-10-26:
Which gives me the impression that the filename should actually end in obj02.cf to avoid all confusion?

tgingold, 2017-10-27:
Not sure it is worth changing the library name.

I agree that this area could be improved, but the design is not here.

closed 2017-10-27 by tgingold