ghdl/ghdl #447: Using alias of generic in function leads to violating pure rule for function.

m-kru, 2017-10-26:
Consider following example:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity my_entity is
    generic (
        GEN: std_ulogic_vector := b"1111"
    );
    port (
        clk: in std_ulogic;
        a: in std_ulogic_vector(7 downto 0);
        b: out std_ulogic_vector(7 downto 0)
    );
end;

architecture behavioral of my_entity is

    alias GEN_DOWNTO: std_ulogic_vector(GEN'length - 1 downto 0) is GEN;

    function foo(x: std_ulogic_vector(7 downto 0))
    return std_ulogic_vector is
        variable ret: std_ulogic_vector(7 downto 0);
    begin
        ret := x;
        for i in natural(GEN_DOWNTO'length - 1) downto 0 loop
            ret(i) := GEN_DOWNTO(i) and x(i);
        end loop;
        return ret;
    end;

begin
    process (clk) begin
        if rising_edge(clk) then
            b <= foo(a);
        end if;
    end process;
end;
```
Running `ghdl -s my_entity.vhd` returns:
```
my_entity.vhd:25:26:warning: reference to alias "gen_downto" violate pure rule for function "foo"
my_entity.vhd:26:23:warning: reference to alias "gen_downto" violate pure rule for function "foo"
```
I am not sure if this warning is proper? Using generics explicitly in functions does not issue any warning.
I have synthesized this simple module in Vivado 2017.3 and Xilinx tools do not issue any similar error or warning.

CraigNoble, 2017-10-26:
It's complaining because you're not passing GEN_DOWNTO in as a parameter, making this a defacto impure function. Also, checkout the REVERSE_RANGE attribute.

`library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity my_entity is
    generic (
        GEN: std_ulogic_vector := b"1111"
    );
    port (
        clk: in std_ulogic;
        a: in std_ulogic_vector(7 downto 0);
        b: out std_ulogic_vector(7 downto 0)
    );
end;

architecture behavioral of my_entity is

    alias GEN_DOWNTO: std_ulogic_vector(GEN'REVERSE_RANGE) is GEN;

    function foo(x: std_ulogic_vector(7 downto 0); GEN_DOWNTO: std_ulogic_vector)
    return std_ulogic_vector is
        variable ret: std_ulogic_vector(7 downto 0);
    begin
        ret := x;
        for i in natural(GEN_DOWNTO'length - 1) downto 0 loop
            ret(i) := GEN_DOWNTO(i) and x(i);
        end loop;
        return ret;
    end;

begin
    process (clk) begin
        if rising_edge(clk) then
            b <= foo(a, GEN_DOWNTO);
        end if;
    end process;
end;`

tgingold, 2017-10-26:
I think that m-kru is correct: the vhdl file is correct.  I am currently fixing that issue.

commit 254b3b89 (2017-10-26): Follow object aliases while checking purity.

Fix #447
closed 2017-10-26 by tgingold