ghdl/ghdl #449: Problems with Overload resolution

go2sh, 2017-10-27:
I did some test with the overload resolution to understand what the lrm means with a `complete context` and so on. My testcases gave me some strange results in various simulators. To examples:
```
entity asd is
end entity;
architecture aa of asd is
type ia is array (natural range <>) of integer;

function a return ia is
begin
return (1,2);
end function
function a (s : integer) return ia is
begin
return (1,2);
end function;

signal ad : integer;
begin
ad <= a(1);
end architecture;
```
GHDL dose analyze this with out complaining. Modelsim tells me `no feasible entries. On the other hand:
```
entity asd is
end entity;

architecture aa of asd is
type ia is array (natural range <>) of integer;

function a return ia is
begin
return (1,2);
end function;

function a (s : integer) return integer is
begin
return 1;
end function;

signal ad : integer;
begin
ad <= a(1);
end architecture;
```
GHDL is complaining now and Modelsim analyzes it.

I think from an LRM point of view GHDL is right in both cases. But I'am not sure.

Brian-Drummond, 2017-10-27:
Your first example appears to have a missing semicolon at 'end function[;]`, I assume this is a copy/paste artefact.

I agree with your conclusions, I have seen similar overload problems with functions with default integer arguments, returning strings, where  `"hello " & f(5)` either returns the fifth string, or the fifth character of the default string (and '&' is overloaded between string and character). Can't find the discussion online offhand, (I think it was on ghdl-discuss@gna.org around December 2014) but IIRC the conclusion then was that ghdl was correct while Modelsim arbitrarily picked one of the possible overloads.

go2sh, 2017-10-27:
Yeah, its a copy and past error.

Especially with the first one: I don't understand, why Modelsim is complaining. Since the overload resolution happens within a compete context. It has to analyze all possibilities and there is only one.

And just for information `innermost complete context` means, that overload resolution is stopped a context complete bound  (e.g signal assignment statement). Especially overload resolution is stopped a a type conversion, even it is part of a statement. Am I correct?

tgingold, 2017-11-02:
go2sh: yes you are correct about innermost complete context.

You should ask ModelSim support about your examples!

tgingold, 2017-11-02:
I am closing this issue, as there is no action to do.
I would be interested if you get an answer from ModelSim.

closed 2017-11-02 by tgingold