ghdl/ghdl #450: Access module in generate statement via VPI

marph91, 2017-10-30:
Hello,
I have a problem to get the correct module names from a generate statement when using cocotb + ghdl (see also [https://stackoverflow.com/questions/46988237/unusable-module-names-with-generate-statement-in-cocotb](url) ).

Using the attached example (modified from [http://ghdl.free.fr/site/uploads/Main/vpi-ex-ghdl.tar](url) ). The output from `vpiFullName` is as expected, but the output from `vpiName` is `(1)` instead of `gen_one(1)`. This seems to be the case for "for generate statements" as well as "if generate statements".

VPI example: [vpi-ex-ghdl.tar.gz](https://github.com/tgingold/ghdl/files/1428446/vpi-ex-ghdl.tar.gz)

tgingold, 2017-10-31:
The vpiName issue is now fixed.  I am not yet sure the cocotb testcase would pass however.

closed 2017-10-31 by tgingold

marph91, 2017-10-31:
Thank you for the fast fix. The "for generate statement" works now. However im still getting `B1` instead of `gen_if` for the "if generate statement" (last two output lines).

tgingold, 2017-10-31:
> On 31 Oct 2017, at 14:59, marph91 <notifications@github.com> wrote:
> 
> Thank you for the fast fix. The "for generate statement" works now. However im still getting B1 instead of gen_if for the "if generate statement" (last two output lines).

You're right.  Now fixed (I hope).

Tristan.

commit 46af8fa0 (2017-10-31): Fix VhpiNameP for VhpiForGenerateK: add generate label.

Fix #450