ghdl/ghdl #451: Argument list in function call should be mandatory

vicencb, 2017-11-05:
Hi,
the LRM says that a subprogram with a default value for all its parameters can be called with or without an association list, but not with an empty one. So, making the association list mandatory would be non-compliant.
But at least it would be nice if a function call allowed an empty association list. This way one could write less confusing code and avoid bugs like this:

When compiling the following code with `ghdl -a --std=08 e.vhd`:
```vhdl
entity e is end entity;
architecture h of e is
  type a is array(natural range<>, natural range<>) of integer_vector;
  function f(x, y:integer := 0) return a is begin return (0 to 3=>(0 to 1=>(0 to 6=>y))); end function;
  -- If the association list was mandatory f()(2,0)(5) would be far less confusing.
  constant c :integer := f(2,0)(5);

  type b is array(natural range<>) of integer_vector;
  function g(x:integer := 0) return b is begin return (0 to 3=>(0 to 6=>45)); end function;
  -- If the association list was mandatory g()(2)(5) would be far less confusing.
  constant d :integer := g(2)(5);
begin
  assert false report integer'image(c) severity note;
  assert false report integer'image(d) severity note;
end architecture;
```
ghdl crashes with the following report:
```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-374-g5e534966) [Dunoon edition]
Compiled with GNAT Version: 7.2.0
Target: x86_64-pc-linux-gnu
In directory: /tmp/compiler/
Command line:
/usr/bin/ghdl1-llvm --std=08 -P/usr/lib/ghdl/v08/std/ -P/usr/lib/ghdl/v08/ieee/ -c -fpic -o e.o e.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : iirs_utils.adb:1085
Call stack traceback locations:
0x5616a9f3f631 0x5616a9dc2e4a 0x5616a9dc2ecd 0x5616a9f02a09 0x5616a9f10571 0x5616a9ecf412 0x5616a9e73ae2 0x5616a9e8cff2 0x5616a9e8d1d5 0x5616a9f16eac 0x5616a9eb5183 0x5616a9e9653f 0x5616a9f21663 0x5616a9e638f6 0x5616a9f27651 0x5616a9f27e48 0x5616a9f29794 0x5616a9d0357e 0x7f006e6edf68 0x5616a9d02978 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

Regards,
&nbsp;&nbsp;Vicenç.

vicencb, 2017-11-05:
Please, also consider the overloaded variant by renaming `g` to `f`.

tgingold, 2017-11-05:
Arrays of unbounded types are not yet supported.

tgingold, 2021-09-07:
Now supported and working.

closed 2021-09-07 by tgingold