ghdl/ghdl #453: llvm crash when compiling uvvm

fjaelltrollet, 2017-11-05:
Hi

If I run compile-uvvm.sh with UVVM version 2.4.1, both ghdl-llvm and ghdl-mcode work fine.
If I run that script with UVVM version v2017.10.16, ghdl-mcode works fine but ghdl-llvm crashes for several files.

One example of the crash message:
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-374-g5e534966-dirty) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
In directory: /home/nevyn/Documents/Freizeit/Stromer/hdl/uvvm/test/uvvm_util/v08/
Command line:
/opt/ghdl/install-llvm/bin/ghdl1-llvm -fexplicit -frelaxed-rules --no-vital-checks --warn-binding --mb-comments --std=08 --work=uvvm_util -P/home/nevyn/Documents/Freizeit/Stromer/hdl/uvvm/test/./ -P/opt/ghdl/install-llvm/lib/ghdl/v08/std/ -P/opt/ghdl/install-llvm/lib/ghdl/v08/ieee/ -c -fpic -o methods_pkg.o /home/nevyn/Documents/Freizeit/Stromer/hdl/uvvm/git/uvvm_util/src/methods_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1621
Call stack traceback locations:
0x58fdf5 0x5eecc9 0x5efdbc 0x5f1e50 0x5f1e31 0x5f238f 0x5f1343 0x5f26b5 0x5b4936 0x5cca15 0x5ccbf6 0x5f7e54 0x5dbb97 0x5f91cf 0x5937f5 0x60aad2 0x60b288 0x48ab03 0x4451fd 0x7f1b577ec2af 0x444618 0xfffffffffffffffe
******************************************************************

Best regards,
emanuel

tgingold, 2017-11-06:
uvvm is now using array of std_logic_vector; this is not yet supported.

svenn71, 2017-11-16:
I would like to use the latest version of uvvm with ghdl and experience this issue. Is array of std_logic_vector something which should go on the roadmap?

tgingold, 2017-11-16:
> On 16 Nov 2017, at 07:36, svenn71 <notifications@github.com> wrote:
> 
> I would like to use the latest version of uvvm with ghdl and experience this issue. Is array of std_logic_vector something which should go on the roadmap?

Yes, this is the next vhdl-2008 feature to be implemented.

tgingold, 2018-11-22:
UVVM can now be built.

closed 2018-11-22 by tgingold