ghdl/ghdl #455: writeline is deleting the line

react66, 2017-11-08:
I was trying to read a file line by line and output to the console as I went along, but writeline deleted the contents. This is a pretty common thing to do so I don't think GHDL should be deleting the contents.

Attaching a testcase, but the following shows the problem:
process 
variable l:line;
begin
write(l,string'("hello world"));
report " I will now output the string";
writeline(output,l);
report " I will now output the string again";
writeline(output,l);
report "did it work?";
wait;
end process;

The output:
readwrite_tst.vhd:14:1:@0ms:(report note):  I will now output the string
hello world
readwrite_tst.vhd:16:1:@0ms:(report note):  I will now output the string again

readwrite_tst.vhd:18:1:@0ms:(report note): did it work?
[ghdl_writeline_issue.zip](https://github.com/tgingold/ghdl/files/1455741/ghdl_writeline_issue.zip)

Paebbels, 2017-11-08:
GHDL is correct in that behavior. It is how the `writeline` procedure is defined in `IEEE.TextIO` and IEEE Std. 1076-2008. Actually, no content is deleted, but the access type to the original content is cleared.

You need to copy your string if you need it twice.

react66, 2017-11-08:
You are correct, I just read it in the standard. My mistake, I was looking at examples. If I copy
m:=l then try to writline m then I get a crash.
../ghdl_writeline_issue=>readwrite_tst
readwrite_tst.vhd:14:1:@0ms:(report note): I will now copy the string pointer
readwrite_tst.vhd:16:1:@0ms:(report note):  I will now output the line m
hello world
readwrite_tst.vhd:18:1:@0ms:(report note):  I will now output the line l
*** Error in `readwrite_tst': double free or corruption (fasttop): 0x00000000016a04e0 ***
hello worldAborted (core dumped)

So how do I copy the line? Note in the real world I am reading the line in from a file,
not just creating strings.

Paebbels, 2017-11-08:
Type `line` As an access type (pointer) to a string (array of `character`). By assigning one access value to another one, you copy references (pointers), no data. Thus, calling `writeline` twice tries to `deallocate` "free" the string value two times. So you need to copy the string content itself.

    m := new string'(l.all);

react66, 2017-11-08:
Lets close the issue, I found that too. Not in the LRM(?) Is the ADA reference our fallback?

closed 2017-11-08 by react66