ghdl/ghdl #456: Exception CONSTRAINT_ERROR raised

gnumbs, 2017-11-10:
Hi,

I'm trying to analyze/elaborate/simulate a VHDL project composed by some files.
One of these, is generating this error.

This file is really big (~12GB for 500.000 rows); it's not an architecture file but a package, containing a matrix and notghing else.
Files with smaller matrices (and then smaller file size and rows) but with the same identical structure, works.

`******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.35-dev (2017-03-01-395-g9a90393e) [Dunoon edition]
Compiled with GNAT Version: 7.2.1 20171025
Target: x86_64-linux-gnu
In directory: /home/ferrandino/Desktop/AES/
Command line:
ghdl -a --work=my_lib --workdir=my_lib structure_1.vhd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : files_map.adb:827 range check failed
Call stack traceback locations:
0x55b1eeb85c5a 0x55b1eeb987e8 0x55b1eed20c11 0x55b1eed10f3b 0x55b1eed3cb88 0x55b1eea555ff 0x7fcfe80712df 0x55b1eea54428 0xfffffffffffffffe
******************************************************************`

tgingold, 2017-11-10:
Can you post a file with a very small matrix ?

gnumbs, 2017-11-11:
Yes, this is a structure file that I used in another project (that works). The only changes are the constant value and clearly the resulting matrix.
Thank you.

`library IEEE;
use IEEE.STD_LOGIC_1164.all;

package crossbar_structure_1 is

constant cb_height : integer := 6;
constant cb_width : integer := 8;

type matrix_row is array (0 to cb_width-1) of integer;
type matrix is array (0 to cb_height-1) of matrix_row;

constant cb_structure : matrix := (
					(1,1,1,1,1,1,0,0),
					(0,1,1,0,0,1,0,1),
					(1,0,0,1,0,1,0,1),
					(0,1,0,0,1,0,0,1),
					(0,0,1,0,1,0,0,1),
					(0,0,0,0,0,0,2,1)
);
end crossbar_structure_1;`

Paebbels, 2017-11-11:
@tgingold 
Is it a parser problem in reading to much data or a simulator problem in handling big data structures?

@gnumbs 
Is it possible to calculate the matrix in a function that is used to initialize the constant?

gnumbs, 2017-11-11:
@Paebbels 
The matrix, and all the others VHDL files of the projects, are written by a C++ tool, on the base of an input file passed to the tool.

diogratia, 2017-11-12:
@paebbels -

commit a90393 files_map.adb 827:

```ada
   --  Return an entry for a filename.
   --  Load the filename if necessary.
   function Read_Source_File (Directory : Name_Id; Name: Name_Id)
                              return Source_File_Entry
   is
      use GNAT.OS_Lib;
      Fd : File_Descriptor;

      Res: Source_File_Entry;

      Length: Source_Ptr;
      Buffer: File_Buffer_Acc;
   begin
      --  If the file is already loaded, nothing to do!
      Res := Find_Source_File (Directory, Name);
      if Res /= No_Source_File_Entry then
         pragma Assert (Source_Files.Table (Res).Source /= null);
         return Res;
      end if;

      --  Open the file (punt on non regular files).
      declare
         Filename : String := Get_Pathname (Directory, Name, True);
      begin
         if not Is_Regular_File (Filename) then
            return No_Source_File_Entry;
         end if;
         Fd := Open_Read (Filename'Address, Binary);
         if Fd = Invalid_FD then
            return No_Source_File_Entry;
         end if;
      end;

      Res := Create_Source_File_Entry (Directory, Name);

      Length := Source_Ptr (File_Length (Fd));
```
Where the last line shown above is 827.

types.ads: 
```ada
type Source_Ptr is new Uns32;
```
File_Length (Fd) is returning a value out of range for Source_Ptr.

ON OS X using Apple's ld64 it would also be impossible to initialize an object  during initialization using greater that 32 bit offset arithmetic. (An object size limit of 4GB, ya, I experimented with initialized a large matrix type with the value of cb_structure initialized to `(others => others => 0))`). It's not clear it would be useful to have a 64 bit Source_Ptr. The hanky pushes out of your fist elsewhere, no magic trick. Sorry, Umberto.

In any event a design file size limit of 4GB doesn't sound unreasonable. I half finished a small C program for exceeding the file size limit to check other implementations using a handy array of 2049 small integer literals I had lying around but didn't know the literal sizes or cb_width being used in issue report. I saw Umberto was using tabs instead of spaces in his C++ VHDL generator, but he does describe the file size as exceeding 12 GB.

For such a large data set driving a model either a FOREIGN function call providing a window on a large dataset in a file returning  matrix_row elements or TEXTIO doing the same could be used.

As long as you aren't communicating between processes (IEEE Std 1076-2008 Annex D Potentially nonportable constructs) TEXTIO should be portable between VHDL implementations. 

For simulation expression value size should be well below 4GB. You'll run out of swap or virtual memory. There are similar limitations on synthesis, let's see you reduce a 4GB ROM in any reasonable time. Our extant computer architectures, hardware description languages and software applications aren't up to the task. We likely need paradigm shifts in all three that would obsolete VHDL (or SystemVerilog, ...).

Paebbels, 2017-11-12:
So we could catch the exception beforehand by checking if an input file exceeds 4 GiB on disk.

Of cause this wouldn't solve the OP's issue.

@gnumbs 
Could you split the constant into 3 VHDL files, each of 4 GiB max and concatenate them later?

tgingold, 2017-11-13:
I have improved the output in case of very large input file.

But yes, your file is way too large.

As hinted by @diogratia, you should generate a data file from C++ and read it using TEXTIO.
I am pretty sure you read your array sequentially, so there is no real need to have an array: just read the inputs from a file.

I am closing this issue, because changing the input is the real solution.  Even if ghdl was able to read very large file (12GB !), it will generate a huge internal tree (the AST could be ~100GB).  I doubt that you have enough memory (I don't have that much, so I won't be able to do any test).

Feel free to continue the discussion if you have any issue with  moving to TEXTIO.

Tristan.

closed 2017-11-13 by tgingold