ghdl/ghdl #457: PowerShell installation failed

josyb, 2017-11-12:
The .ps1 PowerShell script failed on my WIn10 machine.
This what I changed to get it installed:
```
# writing ZIP file to disk
	$TempFilePath =									[System.IO.Path]::GetTempFileName() | Rename-Item -NewName { $_ -replace 'tmp$', 'zip' } –PassThru
	Write-Host "  Writing temporary ZIP file: $TempFilePath"
	$CompressedFileContentAsBytes =	[System.Convert]::FromBase64String($CompressedFileContentInBase64)
	[System.IO.File]::WriteAllBytes("$TempFilePath", $CompressedFileContentAsBytes)
	
	Write-Host "  Extracting ZIP file $TempFilePath to: $InstallPath"
	Expand-Archive -Path $TempFilePath -DestinationPath $InstallPath -Force
```

Paebbels, 2017-11-12:
What was the error message you got before?

I have installed GHDL on different Win7 and Win10 machines without problems.

josyb, 2017-11-12:
```
C:\Users\josy\Downloads\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 : Cannot validate argument on parameter 'AddToPath'.
The argument "C:\Programs\GHDL" does not belong to the set "Machine,User,Session,Remove,Pass" specified by the
ValidateSet attribute. Supply an argument that is in the set and then try the command again.
At line:1 char:53
+ .\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 -Install C:\Programs\GHDL
+                                                     ~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (:) [ghdl-v0.34-gnatgpl32-mcode.installer.ps1], ParameterBindingValidationE
   xception
    + FullyQualifiedErrorId : ParameterArgumentValidationError,ghdl-v0.34-gnatgpl32-mcode.installer.ps1
```
I then added ```-AddtoPath Machine``` but then got:
```
================================================================================
Installing GHDL 0.34-dev for Windows...
  Install directory: C:\Programs\GHDL
  Creating directory 'C:\Programs\GHDL' and sub-directories...
  Writing temporary ZIP file: C:\Users\josy\AppData\Local\Temp\tmp9449.tmp
  Extracting ZIP file to: C:\Programs\GHDL
Expand-Archive : A parameter cannot be found that matches parameter name 'OutputPath'.
At C:\Users\josy\Downloads\ghdl-v0.34-gnatgpl32-mcode.installer.ps1:214 char:33
+     Expand-Archive "$TempFilePath" -OutputPath $InstallPath -Force -S ...
+                                    ~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [Expand-Archive], ParameterBindingException
    + FullyQualifiedErrorId : NamedParameterNotFound,Expand-Archive
```

and then:
``` 
PS C:\Users\josy\Downloads> .\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 -AddToPath Machine -Install "C:\Programs\GHDL"
================================================================================
GHDL for Windows - GHDL install script
================================================================================
Installing GHDL 0.34-dev for Windows...
  Install directory: C:\Programs\GHDL
  Creating directory 'C:\Programs\GHDL' and sub-directories...
  Writing temporary ZIP file: C:\Users\josy\AppData\Local\Temp\tmp28F.tmp
  Extracting ZIP file C:\Users\josy\AppData\Local\Temp\tmp28F.tmp to: C:\Programs\GHDL
Expand-Archive : .tmp is not a supported archive file format. .zip is the only supported archive file format.
At C:\Users\josy\Downloads\ghdl-v0.34-gnatgpl32-mcode.installer.ps1:214 char:2
+     Expand-Archive -Path $TempFilePath -DestinationPath $InstallPath  ...
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (.tmp:String) [Expand-Archive], IOException
    + FullyQualifiedErrorId : NotSupportedArchiveFileExtension,Expand-Archive
```

josyb, 2017-11-12:
This is the complete successful command:
```
PS C:\Users\josy\Downloads> .\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 -AddToPath Machine -Install "C:\Programs\GHDL"
================================================================================
GHDL for Windows - GHDL install script
================================================================================
Installing GHDL 0.34-dev for Windows...
  Install directory: C:\Programs\GHDL
  Creating directory 'C:\Programs\GHDL' and sub-directories...
  Writing temporary ZIP file: C:\Users\josy\AppData\Local\Temp\tmp5CDD.zip
  Extracting ZIP file C:\Users\josy\AppData\Local\Temp\tmp5CDD.zip to: C:\Programs\GHDL
  Adding GHDL to PATH at machine level.

Installing files [SUCCESSFUL]
``

Paebbels, 2017-11-12:
OK. Now I see the problem.

> Expand-Archive : .tmp is not a supported archive file format. .zip is the only supported archive file format.

It might be possible that one of my installed PowerShell extensions (PSCX) accepts more file extensions than on a regular Win10 setup.

The result of `[System.IO.Path]::GetTempFileName()` is a string.
So adding a simple `-replace "tmp$","zip"` should be enough.

Btw. it's not correct to install GHDL into `C:\Programs`.
GHDL is a x86 binary not a x86-64 binary. The default directory is chosen if no parameter is supplied.

josyb, 2017-11-12:
>Btw. it's not correct to install GHDL into C:\Programs.

I installed in "C:\Programs" which is not "C:\Program Files". I apologize if I confused you.

josyb, 2017-11-12:
> So adding a simple -replace "tmp$","zip" should be enough.

I needed to add the ```-AddToPath Machine``` as well.

Paebbels, 2017-11-12:
The option `AddToPath` is needed if you don't want to get a question about how to add GHDL to `$env:Path`.

If you supply an answer as parameter, no question is asked. Otherwise, the script asks for a level where to modify `$env:Path`. The option is not required: `Mandatory=$false`.

If your machine behaves differently, please post the error message for a missing `AddToPath` parameter.

josyb, 2017-11-12:
This is what happened on my very first invocation:
```
PS C:\Users\josy\Downloads> .\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 -Install C:\Programs\GHDL

Security warning
Run only scripts that you trust. While scripts from the internet can be useful, this script can potentially harm your
computer. If you trust this script, use the Unblock-File cmdlet to allow the script to run without this warning
message. Do you want to run C:\Users\josy\Downloads\ghdl-v0.34-gnatgpl32-mcode.installer.ps1?
[D] Do not run  [R] Run once  [S] Suspend  [?] Help (default is "D"): R
C:\Users\josy\Downloads\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 : Cannot validate argument on parameter 'AddToPath'.
The argument "C:\Programs\GHDL" does not belong to the set "Machine,User,Session,Remove,Pass" specified by the
ValidateSet attribute. Supply an argument that is in the set and then try the command again.
At line:1 char:53
+ .\ghdl-v0.34-gnatgpl32-mcode.installer.ps1 -Install C:\Programs\GHDL
+                                                     ~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (:) [ghdl-v0.34-gnatgpl32-mcode.installer.ps1], ParameterBindingValidationE
   xception
    + FullyQualifiedErrorId : ParameterArgumentValidationError,ghdl-v0.34-gnatgpl32-mcode.installer.ps1
```

Paebbels, 2017-11-12:
Thank you very much. That was very help full!

The installation path `InstallDir` is set to be parsed - if not explicitly set - as last parameter. Unfortunately, PowerShell tries to parse the provided installation path as `AddtToPath`, which has a verification set: `machine, user, session, ...`

I have to figure out, how to solve it ....
I would like to allow auto completion for `AddToPath`. That's why I used a validation set. I could do validation on my own and check if a parameter is an installation path or paramter for `AddToPath`, but then auto completion is not possible anymore.

I'm open for suggestions.

I would like to support these inputs:

```PowerShell
.\installer.ps1 -Install
.\installer.ps1 -Install -AddToPath Machine
.\installer.ps1 -Update
.\installer.ps1 -Update -AddToPath Machine
.\installer.ps1 -Install C:\Tools\GHDL\0.34
.\installer.ps1 -Install C:\Tools\GHDL\0.34 -AddToPath Machine
.\installer.ps1 -Install -AddToPath Machine C:\Tools\GHDL\0.34
# This is still possible, because every parameter can be passed by name
.\installer.ps1 -Install -InstallDir C:\Tools\GHDL\0.34 -AddToPath Machine
.\installer.ps1 -Install -AddToPath Machine -InstallDir C:\Tools\GHDL\0.34
```

I'm thinking, maybe it's possible to combine the parameters `Install` and `InstallDir`. But this requires that the parameter accepts `$false` and a `String` or `$null` if this is possible.

--------------
Regarding the security warning:

> **Security warning**
> Run only scripts that you trust. While scripts from the internet can be useful, this script can potentially harm your computer. If you trust this script, use the Unblock-File cmdlet to allow the script to run without this warning message.

By default, PowerShell requires signed scripts from everywhere:
* local computer / self-written or
* downloaded / from internet / from intranet.

You can lower the boundary by using `Set-ExecutionPolicy` to only required signed scripts if the where downloaded or executed from a remote source:

```PowerShell
Set-ExecutionPolicy RemoteSigned
```

josyb, 2017-11-13:
I dealt with that warning as you indicated

honzucha, 2017-12-29:
According the Extract-Archive issue, why not use system Expand-Archive as in plain install of win10

I would suggest to change line 208 to
$TempFilePath = "$([System.IO.Path]::GetTempFileName()).zip"

and   214 to 
Expand-Archive "$TempFilePath" -DestinationPath $InstallPath -Force 

problem solved and installed successfully for me.

tested on Win10 PS version 5.1 Build 16299 Revision 98

Kind Regards..

Paebbels, 2017-12-29:
@honzucha This issue is not about `Expand-Archive`. It is about option `-AddToPath`.

The only difference that I see is the changed file name? How does it solve the problem?
What was the error message on your Windows 10 installation?

honzucha, 2017-12-29:
It is about expand-archive, while the system one version doesn't support OutputPath and ShowProgress parameters.

see https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.archive/expand-archive?view=powershell-5.1

error message as follows
================================================================================
GHDL for Windows - GHDL install script
================================================================================
Installing GHDL 0.34-dev for Windows...
  Install directory: -InstallPath C:\GHDL
  Creating directory '-InstallPath C:\GHDL' and sub-directories...
  Writing temporary ZIP file: C:\Users\me\AppData\Local\Temp\tmpE74A.tmp
  Extracting ZIP file to: -InstallPath C:\GHDL
Expand-Archive : A parameter cannot be found that matches parameter name 'OutputPath'.
At C:\Users\me\Downloads\ghdl-v0.34-gnatgpl32-mcode.installer.ps1:214 char:33
+     Expand-Archive "$TempFilePath" -OutputPath $InstallPath -Force -S ...
+                                    ~~~~~~~~~~~
    + CategoryInfo          : InvalidArgument: (:) [Expand-Archive], ParameterBindingException
    + FullyQualifiedErrorId : NamedParameterNotFound,Expand-Archive

Paebbels, 2017-12-29:
@honzucha No, this issue (#457) started by @josyb is about `-AddToPath`.  
Your issue is about `Expand-Archive`.

-----------------------------
Anyhow, I see the problem.  
My system has 2 `Expand-Archive` commands:
* from Microsoft.PowerShell.Archive (1.0.1.0)
* from Pscx (3.2.2)

---------------------
Related Stackoverflow question: [How to call an overloaded command in Powershell](https://stackoverflow.com/questions/48024629/how-to-call-a-overloaded-command-in-powershell)

honzucha, 2017-12-29:
@Paebbels  exactly and I would prefer not to use 3rd party commandlets as it is giving another dependencies.
@josyb mentioned this also in one of his posts from Nov 12 than one with 3 error messages. in the middle after "I then added -AddtoPath Machine but then got:" ...
That's the reason why I see this related.

Anyway after making those 2 suggested changes install script was working as a charm for me.

another suggested change, I would call it cosmetic change..
While PS is also ported to OS X and Linux I would add Platform Check to the beginning of the script

something like 

if ([environment]::OSVersion.Platform -ne "WIN32NT") {Write-Error "Powershell Installer not supported on Linux/Unix Platforms"}

but this is for another "ticket".

Anyway thx for your great job with GHDL...

Paebbels, 2017-12-29:
Yes, for end users, it's OK to have less dependencies, but the script should still be executable on my computer and on any other computer that uses common extensions like Pscx unless PowerShell adopts all these nice additions, that were collected since PowerShell 2.0 was not so powerful as today.

Suggested solution form Stackoverflow: call the command with FQN

```powershell
Microsoft.PowerShell.Archive\Expand-Archive ....
```

Btw. you error messages reports GHDL 0.34. We already releases 0.35 also as a PowerShell installer.

honzucha, 2017-12-29:
Yep, I tested those changes on 0.35... 
FQN will work, there is ticket https://github.com/Pscx/Pscx/issues/23 about conflicts with native PS 5.1 commands. In general external tools should not conflict with native ones... but the pcsx Expand-Archive was the first. another solution (from my point cleaner one) is importing Pscx module with prefix

Import-Module Pscx -Prefix Pscx
Expand-PscxArchive 

as suggested on ticket by rkeithhill.

Paebbels, 2018-11-11:
This issue should be fixed by #681.

closed 2018-11-11 by Paebbels

houseofwealth, 2018-12-04:
I am getting the same error when I try the installer:
```
..> .\ghdl-0.35-mcode-windows.installer.ps1 -Install c:\ProgramFiles\GHDL-0.35
..\ghdl-0.35-mcode-windows.installer.ps1 : Cannot validate argument on parameter
'AddToPath'. The argument "c:\ProgramFiles\GHDL-0.35" does not belong to the set "Machine,User,Session,Remove,Pass" specified by the ValidateSet attribute. Supply an argument that is in the set and then try the command again.
At line:1 char:50
+ ... l-0.35-mcode-windows.installer.ps1 -Install c:\ProgramFiles\GHDL-0.35
+                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (:) [ghdl-0.35-mcode-windows.installer.ps1], ParameterBindingValidationExce
   ption
    + FullyQualifiedErrorId : ParameterArgumentValidationError,ghdl-0.35-mcode-windows.installer.ps1

```
thanks

Paebbels, 2018-12-04:
Have you downloaded the latest package file for v0.35?

houseofwealth, 2018-12-04:
Sorry which file is that? I only downloaded the installer

Paebbels, 2018-12-04:
This file: https://github.com/ghdl/ghdl/releases/download/v0.35/ghdl-0.35-mcode-windows.installer.ps1
Was it downloaded after I fixed #681 ?

houseofwealth, 2018-12-04:
I downloaded it just an hour ago.