ghdl/ghdl #458: Attribute 'delayed has a rejection limit

Paebbels, 2017-11-14:
The attribute `'delayed(<time>)` has a rejection limit greater than 0 fs.
According to LRM "16.2.4 Predefined attributes of signals", the attribute usage `S'delayed(T)` is transformed to an equivalent process:

```VHDL
P: process (S)
begin
  R <= transport S after T;
end process;
```

Here is a reproducer that checks all three delay mechanisms:
```VHDL
entity E is
end entity;

architecture A of E is
	signal S1           : bit := '0';
	signal S2_inertial  : bit;
	signal S2_transport : bit;
	signal S2_delayed   : bit;
	
	constant LEVEL : severity_level := ERROR; -- FAILURE;
begin
	S1 <= '1' after 10 ns, '0' after 20 ns;
	
	S2_inertial  <= inertial S1 after 100 ns;
	S2_transport <= transport S1 after 100 ns;
	S2_delayed   <= S1'delayed(100 ns);
	
	CheckInertial: process
	begin
		wait until S2_inertial = '1' for 200 ns;
		assert (S2_inertial = '0') report "Pulse was not rejected!" severity LEVEL;
		wait;
	end process;

	CheckTransport: process
	begin
		wait until S2_transport = '1' for 115 ns;
		assert (S2_transport = '1') report "Pulse was not transport delayed!" severity LEVEL;
		assert (now = 110 ns)       report "Transport delayed pulse was not received at 110 ns!" severity LEVEL;
		wait;
	end process;
	
	CheckDelayed: process
	begin
		wait until S2_delayed = '1' for 115 ns;
		assert (S2_delayed = '1') report "Pulse was not delayed!" severity LEVEL;
		assert (now = 110 ns)     report "Delayed pulse was not received at 110 ns!" severity LEVEL;
		wait;
	end process;
end architecture;
```

While the transport delayed signal works as expected, the `delayed` signal was rejected.

The bug wrong behavior can be shown with VHDL-93, -2002 and -2008.

-------------------
Tested with GHDL 0.35-dev (mcode) on Windows 10:
```
GHDL 0.35-dev (commit: 2017-11-08;  git branch: paebbels/master';  hash: b9fd5ff3) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 mcode code generator
Written by Tristan Gingold.
```

Paebbels, 2017-11-14:
Here is a follow up question: **Is the rejection limit *exclusive* or *inclusive*?**

So what about?
```VHDL
S2_reject <= reject 10 ns inertial S1 after 100 ns;
```
Currently, the 10 ns pulse is filtered. I think this is false, too.

Let's take these equivalent lines:
```VHDL
S2_inertial1 <= S1 after 10 ns;
S2_inertial2 <= inertial S1 after 10 ns;
S2_inertial3 <= reject 10 ns inertial S1 after 10 ns;
```
In these three case, a 10 ns pulse is not filtered.
But if I use `reject 10 ns inertial S1 after 100 ns` a 10 ns pulse get's filtered.

---------------------------
Notification list: @JimLewis

JimLewis, 2017-11-14:
During a VHDL simulation cycle / delta cycle, first all signals are updated, then any code that needs to be executed is executed.   From this and the specifics of your signal changes, you should be able to deduce correct behavior.

Paebbels, 2017-11-15:
They question is more about how rejection itself works.

I specified the same rejection time of 10 ns, but I see different behavior if the delay given by *after* is also 10 ns or greater 10 ns (in my case 100 ns).

As far as I have understood rejection, every scheduled transaction until the rejection time is deleted. Looking at the default case and at `reject 10 ns`, then the operation is an **exclusive** operation.

Taken this, `reject 10 ns inertial S1 after 100 ns` is broken too.

Here are no additional delta cycles involved.

tgingold, 2017-11-15:
There was a bug about the handling of delayed.  Will be fixed shortly.

closed 2017-11-15 by tgingold

Paebbels, 2017-11-15:
Regarding the rejection limit: ModelSim and Riviera-PRO have the same behavior as GHDL. The question is still: Why? I'll create a StackOverflow question.

commit e973f74e (2017-11-15): grt-signals: fix thinko in delayed_implicit_process.

Fix #458